lvgl/lv_port_pc_visual_studio #6: When adding C++ classes we are unable to compile

mvturnho, 2020-03-12:
When adding C++ class files the project can not be compiled. You have to set the project properties to do so by setting compile as to default.

[image]

But then when you compile lvgl is generating compile errors;
[image]

When removing the C++ classes and setting the compiler back to Compile as C Code (/TC) it all compiles and runs fine.
[image]

How can I solve this and combine my C++ classes with lvgl in visual studio?

embeddedt, 2020-03-12:
First, you need to compile the LittlevGL files as C code and your C++ classes as C++ code. I'm not sure if you are doing this already.

Then you will most likely run into the fact that Visual Studio's C++ compiler has [issues](https://stackoverflow.com/questions/33270731/error-c4576-in-vs2015-enterprise) with one of the initializer constructs used in LittlevGL. This issue has been reported before in #2; the solution there was to switch to a different IDE which includes a compiler with support for this construct.

Triangulix, 2020-03-21:
Hi
I would like to use LV in an Arduino project which is C++ but also use the simulator to have a quicker way for creating the UI.

Is there a viable way to use the simulator together with C++ code in Visual Studio? Maybe creating 2 projects and referencing the C project from the C++ project?
Or should I forget the idea and go without the simulator?

Thanks

kisvegabor, 2020-03-24:
Take a look at the [platform_io](https://github.com/littlevgl/lv_platformio/) project. Hope it helps.

Triangulix, 2020-03-24:
Thanks, but I now went for writing my own object oriented GUI system.

kisvegabor, 2020-03-24:
Okay, good luck with your project!

closed 2020-04-23 by stale[bot]

embeddedt, 2020-08-10:
Fixed by https://github.com/lvgl/lvgl/pull/1711 (will be released with 7.4).