lvgl/lv_port_pc_visual_studio #7: Need help update lvgl library to v7.0

summer666lyf, 2020-05-22:
I'm tried to build the example, but find this is lvglv6.0, can your update to v7.0?

embeddedt, 2020-05-22:
See https://github.com/lvgl/lvgl/issues/1483#issuecomment-624429715

It will be updated soon.

MCF, 2020-05-30:
This is in progress. waiting for a change to lvgl to be complete before it can be finished:

https://github.com/lvgl/lvgl/pull/1542

DeonMarais64, 2020-05-31:
https://forum.lvgl.io/t/tft-simulator-on-visual-studio-2017-how-to-change-lvgl-from-v6-x-to-dev7-0/2263/7

summer666lyf, 2020-06-01:
Hi all,
Thank  you for your support. The problem has been solved.

closed 2020-06-01 by summer666lyf

MCF, 2020-06-01:
FWIW the lvgl visual studio compile errors have been fixed and this project has now been updated to work with the latest version of lvgl.

kisvegabor, 2020-06-01:
Awesome! Thank you very much, Mike!