lvgl/lv_port_pc_visual_studio #11: Question: how to run all examples in one run?

DaPa, 2020-07-07:
In order to run all examples one after another I did a hack to deinit/reinit LVGL and SDL (not sure if both were needed to be reset):
- https://github.com/DaPa/lvgl/commit/9135d303534cfe55f481107f84a23ce42ef66183
- https://github.com/DaPa/lv_drivers/commit/325d8048443a8a6c43bfc79ddf623c086765c8af
- https://github.com/DaPa/lv_sim_visual_studio_sdl/commit/67ecd4f3403ddd3e9e66ad443add4191a2b98e8f
Is there a better/safer way to achieve that?
But without deinit the next example was kind of drawing over the previous example's GUI elements...

embeddedt, 2020-07-07:
The examples weren't designed for being deinitialized, but for most of them, I think deleting the active screen and recreating it should be enough.

```c
/* run first example */
example1();
/* clean up first example */
lv_obj_t * new_scr = lv_obj_create(NULL, NULL);
lv_scr_load(new_scr);
lv_obj_del(lv_scr_act());
/* run second example */
example2();
```

DaPa, 2020-07-07:
Thanks for the info. I think I have to handle the tasks in between examples too, right? It might be something else needed as I did try this small example:

```
int main(int argc, char** argv)
{
    bool exit = false;
    /*Initialize LittlevGL*/
    lv_init();

    /*Initialize the HAL for LittlevGL*/
    hal_init();

    /* first example */
    lv_demo_widgets();
    /* handle tasks */
    while (!exit) {
        /* Periodically call the lv_task handler.
        * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        for (int i = 1; i < 10; ++i) {
            if (GetAsyncKeyState(VK_ESCAPE)) {
                exit = true;
                break;
            }
            Sleep(1);       /*Just to let the system breathe */
        }
    }
    /* clean up first example */
    lv_obj_t * new_scr = lv_obj_create(NULL, NULL);
    lv_obj_t * old_scr = lv_scr_act();
    lv_scr_load(new_scr);
    lv_obj_del(old_scr);

    /* second example */
    lv_demo_benchmark();
    while (1) {
        /* Periodically call the lv_task handler.
        * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        Sleep(10);       /*Just to let the system breathe */
    }
    return 0;
}
```

First example runs fine, then after ESC, second example appears but crashes immediately:

> lv_sim_visual_studio_sdl.exe!lv_bar_get_max_value(const _lv_obj_t * bar) Line 318
	at visual_studio_2017_sdl\lvgl\src\lv_widgets\lv_bar.c(318)
lv_sim_visual_studio_sdl.exe!bar_anim(_lv_task_t * t) Line 600
	at visual_studio_2017_sdl\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c(600)
lv_sim_visual_studio_sdl.exe!lv_task_exec(_lv_task_t * task) Line 401
	at visual_studio_2017_sdl\lvgl\src\lv_misc\lv_task.c(401)
lv_sim_visual_studio_sdl.exe!lv_task_handler() Line 142
	at visual_studio_2017_sdl\lvgl\src\lv_misc\lv_task.c(142)
lv_sim_visual_studio_sdl.exe!SDL_main(int argc, char * * argv) Line 124
	at visual_studio_2017_sdl\main.c(124)
lv_sim_visual_studio_sdl.exe!main_getcmdline() Line 71
	at c:\projects\sdl\src\main\windows\sdl_windows_main.c(71)

I'm closing this and don't expect anyone to spend more time with this, with the hack I made it works, was just wondering if it is any other way, and indeed this a corner case...
Thank you again for your help!

closed 2020-07-07 by DaPa