lvgl/lv_port_pc_visual_studio #13: Build error. 224 errors on the newest version of VS 2019

mahyar1284, 2020-07-10:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open source file: 'lv_drivers/display/R61581.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_misc/lv_debug.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_core/lv_disp.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_core/lv_group.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_core/lv_indev.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_core/lv_obj.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_core/lv_refr.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_core/lv_style.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_arc.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_blend.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_img.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_label.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_line.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_mask.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_rect.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_draw_triangle.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_img_buf.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_img_cache.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_draw/lv_img_decoder.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_dejavu_16_persian_hebrew.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_fmt_txt.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_12.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_12_subpx.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_14.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_16.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_18.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_20.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_22.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_24.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_26.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_28.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_28_compressed.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_30.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_32.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_34.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_36.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_38.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lvgl/src/lv_font/lv_font_montserrat_40.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
[... cut ...]
Error	C1083	Cannot open source file: 'lv_examples/src/lv_ex_widgets/lv_ex_tileview/lv_ex_tileview_1.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open source file: 'lv_examples/src/lv_ex_widgets/lv_ex_win/lv_ex_win_1.c': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\c1	1	
Error	C1083	Cannot open include file: 'lvgl/lvgl.h': No such file or directory	lv_sim_visual_studio_sdl	C:\Users\mahya\linux_projects\sim\lv_sim_visual_studio_sdl\visual_studio_2017_sdl\main.c	12

DaPa, 2020-07-10:
When you clone this repo you have to clone it with --recursive flag...

git clone git@github.com:lvgl/lv_sim_visual_studio_sdl.git --recursive

embeddedt, 2020-07-10:
Alternatively, you can download the submodules after the fact with `git submodule update --init --recursive`. Both methods do the same thing; the second one just saves you having to clone it a second time.

mahyar1284, 2020-07-10:
hello everyone and thanks for your support, the issue has been solved. I'm closing this issue.

closed 2020-07-10 by mahyar1284