lvgl/lv_port_pc_visual_studio #14: Error

ddanneels, 2020-12-23:
I've just installed VS 2019 and cloned the simulator under W10. All is compiling and starts successfully but after a few seconds and some clicks here and there, I encounter the following error :
Especially, when I run the "widgets" demo and try to scroll the second tab "Visuals" down to bar and led.
It also occurs with the "printer" demo.

```
Error: _lv_mem_buf_get  (lv_mem.c #539 _lv_mem_buf_get())
Error: No free buffer. Increase LV_MEM_BUF_MAX_NUM. (0x00000000)        (lv_debug.c #127 lv_debug_log_error())
```

embeddedt, 2020-12-23:
It's the second report we've gotten of this: https://forum.lvgl.io/t/memory-and-esp32/4050

@kisvegabor Maybe we really do need to increase the default?

MCF, 2020-12-23:
Happens in visual studio 2017 as well (FWIW).

MCF, 2020-12-23:
As mentioned in the linked forum thread, you can add this line to your lv_conf.h file and it will fix the problem:

```
#define LV_MEM_BUF_MAX_NUM   32
```

kisvegabor, 2020-12-23:
It should be already fixed in the master branch Is it working for you too?

MCF, 2020-12-23:
I've done a quick test, updating to the tip of the current master branch of lvgl will also fix the problem.

I'll update the references to lvgl when there is a new release.  Until then use the workaround I mentioned above.

ddanneels, 2020-12-23:
Thanks. I do confirm that changing lvgl submodule to the current master solved the problem.

closed 2020-12-23 by ddanneels