lvgl/lv_port_pc_visual_studio #15: lowerdown filesize

graysuit, 2020-12-24:
Respected developer,
First of all thanks for porting lvgl to my favorite IDE visual studio
I compiled default demo example and I got 5.26 mb file size even with "**Use mfc in shared lib**" checked.
Do you have any advice or suggestion for reducing file_size ?
Regards.

MCF, 2020-12-27:
I'm not sure I understand your question.  Which file are you referring to?  Why are you trying to minimize its size?

graysuit, 2020-12-28:
I mean the demo executable which is build under main Debug folder. It contains various [examples](https://github.com/lvgl/lv_examples/tree/a3bece773739d9885243efe1998295526db3064c) like buttons,trackbar etc.

> Why are you trying to minimize its size?

Obviously to share executables easily. Our some sides have low net connections. So I try my best to lowerdown its size.

embeddedt, 2020-12-28:
Have you tried zipping it? I think EXE files can compress to some extent. You can also try turning off unneeded features/fonts in `lv_conf.h`.

graysuit, 2020-12-30:
@embeddedt Thanks sir.
I removed unused fonts and it becomes 4.33 mb.
```cpp
//#define LV_FONT_MONTSERRAT_8     1
#define LV_FONT_MONTSERRAT_10    1
#define LV_FONT_MONTSERRAT_12    1
//#define LV_FONT_MONTSERRAT_14    1
#define LV_FONT_MONTSERRAT_16    1
//#define LV_FONT_MONTSERRAT_18    1
#define LV_FONT_MONTSERRAT_20    1
#define LV_FONT_MONTSERRAT_22    1
#define LV_FONT_MONTSERRAT_24    1
//#define LV_FONT_MONTSERRAT_26    1
#define LV_FONT_MONTSERRAT_28    1
#define LV_FONT_MONTSERRAT_30    1
#define LV_FONT_MONTSERRAT_32    1
//#define LV_FONT_MONTSERRAT_34    1
//#define LV_FONT_MONTSERRAT_36    1
//#define LV_FONT_MONTSERRAT_38    1
//#define LV_FONT_MONTSERRAT_40    1
//#define LV_FONT_MONTSERRAT_42    1
//#define LV_FONT_MONTSERRAT_44    1
//#define LV_FONT_MONTSERRAT_46    1
//#define LV_FONT_MONTSERRAT_48    1
```
When I highly compress it using 7zip, it becomes 0.8 mb.
Any other idea to further decrease it ?

embeddedt, 2020-12-30:
Not really; is 800KB still too large for your network?

graysuit, 2020-12-30:
@embeddedt 800 kb is when in compressed form and even its just hello world.
My project will use alot of GUI widgets. so imagine how much size it would become ?

embeddedt, 2020-12-30:
I'm not sure if Visual Studio's linker optimizes out unused widgets or not. It's quite plausible that the binary size won't grow that much.

graysuit, 2020-12-30:
@embeddedt 
Visual studio will surely have ability to be called good shrinking linker.
But here I will try cut off things myself at large scale as you know human linker is better than VS linker. xD
I want to use default system things including fonts. Do you have any advice for me ? 

What I like most about lvgl was its cool widgets.

embeddedt, 2020-12-30:
Here's an idea: you can test disabling some widget types to see if the binary size changes.

graysuit, 2021-01-01:
No change on binary size even if I disable all widgets.
I checked the VS linker settings. Linker can shrink up to 3.4 mb.

MCF, 2021-01-04:
I'm closing this issue because it doesn't seem to be an actual bug with the simulator.  It seems to be more of a support discussion.  If you feel differently you are welcome to re-open.

As a side-note, you may reach a wider audience with a support request in the lvgl forum: https://forum.lvgl.io/

closed 2021-01-04 by MCF