lvgl/lv_port_pc_visual_studio #16: VS2019 ERRO

juneofive, 2021-01-08:
After updating all the submodules, open `lv_sim_visual_studio_sdl.sln` directly, and the following error occurs
[image]

> The first type : file is stored in the wrong directory,
Solution:
copy all files from `...\lv_examples\src\assets` to `...\lv_examples\assets`

>The second  is missing files
The solution:
creat a file named `lv_ex_templ.c` at `lv_examples\src\lv_ex_widgets`
I think the error is caused by some description files, and since I don't know the sln structure of VS well, I can't fundamentally solve it

kisvegabor, 2021-01-09:
The images and fonts were really moved to `lv_examples/src/assets` from `lv_example/assets`. I think it requires only a refresh on the file list. 

@MCF Could you take look at this?

MCF, 2021-01-11:
So I've had a look and am not sure why you are seeing a problem.  I will say that I am still running visual studio 2017 so it is possible that in 2019 it is acting differently - but I can't see why it would have the problems you are describing.  A couple of notes:

1. There is no reference to a lv_ex_templ.c in any of the code.  Not in the visual studio files or in any of the source files.  I don't understand why it would be noted as missing since it is not listed in any of the files.

2. I've checked the references to the files in `...\lv_examples\src\assets\` directory.  In the visual studio files they are correct.  You can check this yourself just look in the `visual_studio_2017_sdl.vcxproj` and `visual_studio_2017_sdl.vcxproj.filters` files.  They are simple xml files and the references are easily read.

MCF, 2021-01-11:
Can you ensure you have pulled in the latest changes and updated your submodules.  If you are cloning from the github.com/lvgl copy of the repository (in other words you don't have a fork) then the following two commands will do that for you:

MCF, 2021-01-11:
Alternatively you could just clone the repository to a new directory (including the submodules) and open the solution up again in visual studio.

juneofive, 2021-01-13:
i have updated all files to newest vesion,it's already resolved this problem

closed 2021-01-13 by juneofive