lvgl/lv_port_pc_visual_studio #21: how to set the screen size?

yanmowudi, 2021-02-22:
the screen size always  is 1420*716

MouriNaruto, 2021-02-22:
The window is resizable. (So it uses default window size.)

But I will provide method to change window size in the recent days.

closed 2021-02-22 by yanmowudi

MouriNaruto, 2021-02-24:
I hope you will like it.
https://github.com/lvgl/lv_sim_visual_studio/blob/54355e6f78884c75e84862df7789037c9f189798/LVGL.Windows.Simulator/LVGL.Windows.Simulator.cpp#L426

yanmowudi, 2021-02-24:
thanks a lot.it works.
[image]

您是在arm架构下编译的工程吗？

MouriNaruto, 2021-02-24:
> 您是在arm架构下编译的工程吗？

我在 Windows 10 Version 20H1 x64 下使用 Visual Studio 2019 进行模拟器的开发
虽然你可以生成 arm 和 arm64 的模拟器二进制，可以在 arm 和 arm64 的桌面版 Windows 使用

yanmowudi, 2021-02-24:
您的工程clone下来，默认的解决方案平台是arm

MouriNaruto, 2021-02-24:
> 您的工程clone下来，默认的解决方案平台是arm

其实是因为按照字母顺序排列……arm 在最上面，于是 Visual Studio 默认就选择了

yanmowudi, 2021-02-24:
谢谢解答 :thumbsup: