lvgl/lv_port_pc_visual_studio #24: Why do I display "Not a valid Win32 application"?

TaXue-DuBa, 2021-02-26:
First,I copied the "lvgl" and "lvgl_examples" into "LVGL.Windows/lvgl" and "LVGL.Windows/lvgl_examples" directories.
Then,I used the Visual Studio 2019 and double click the "LVGL.Simulator.sln" to open the VS2019.
Select the x64 and Local Windows Debugger in the top bar.
The compilation seemed to go smoothly.
Finally,the VS2019 shows “"lv_sim_visual_studio-master\Output\Debug\x64\LVGL.Windows.lib" Not a valid Win32 application".

MouriNaruto, 2021-02-26:
Please set the LVGL.Windows.Desktop project as the startup project.

TaXue-DuBa, 2021-02-26:
Thank you very much for your answer!

This is my first time using VS2019 and I was not aware of the existence of "startup option".
I noticed that your README.md already stated it, but I didn't notice it.
I set "LVGL.Windows.Simulator" as "Start the project ".
It worked perfectly.
I'm really sorry for the trouble caused to you. Thank you very much for recovering me and solving my big problem.
Thank you very much!

closed 2021-02-26 by TaXue-DuBa

MouriNaruto, 2021-02-27:
But I think I will merge the projects in the recent days.

TaXue-DuBa, 2021-03-07:
> But I think I will merge the projects in the recent days.

I think that's great, because then this problem doesn't arise.

MouriNaruto, 2021-03-07:
@TaXue-DuBa You can try the latest commit.