lvgl/lv_port_pc_visual_studio #26: visual_studio Compilation fails

zhuxie99, 2021-04-16:
[image]

I followed the instructions to clone ,and Build &run ,but failed,The error is shown in the figure above

zhuxie99, 2021-04-16:
i use Visual Studio 2019

zhuxie99, 2021-04-16:
[image]
Error message in English environment

MouriNaruto, 2021-04-16:
Which the branch you used?

zhuxie99, 2021-04-16:
I use this command to clone git clone --recurse-submodules https://github.com/lvgl/lv_sim_visual_studio.git

i think master branch

MouriNaruto, 2021-04-16:
@zhuxie99 I had clone the repo to the other folder just now and compiled successfully.
[image]

Here is my command list.

> git clone https://github.com/lvgl/lv_sim_visual_studio
> cd lv_sim_visual_studio
> git submodule update --init --recursive
> BuildAllTargets.cmd

I also wonder to know which Visual Studio version you are using.

zhuxie99, 2021-04-16:
[image]

我按照你的命令重新试一遍，之前是一条命令执行，现在分两条命令在试一下

MouriNaruto, 2021-04-16:
Note: Visual Studio compilation result.
[image]
[image]

MouriNaruto, 2021-04-16:
> [image]
> 
> 我按照你的命令重新试一遍，之前是一条命令执行，现在分两条命令在试一下

我感觉还是老版本 Visual Studio 2019 提供的 MSVC 工具链的警告策略和新版本不同的关系。
倒是我习惯用 /W4 /WX 了，也就是 Level 4 警告且视警告为错误，我感觉可以把 /WX 去掉，也许你就可以编译了

zhuxie99, 2021-04-16:
我之前试过了，还是编译不通过，现在我正在更新vs版本到最新版

MouriNaruto, 2021-04-16:
> 我之前试过了，还是编译不通过，现在我正在更新vs版本到最新版

去除 /WX 的事情最好直接在 vcxproj 修改，毕竟我对 vcxproj 进行了精简，否则要同时支持四个架构（x86 x64 arm32 arm64）各两种配置（Debug Release）会比较乱

zhuxie99, 2021-04-16:
你那边如果把 /WX开启，是否会出现这些警告和错误

zhuxie99, 2021-04-16:
感谢，我把vs升级到最新版本，可以编译通过了，能正常运行了

MouriNaruto, 2021-04-23:
v8 的适配分支已经使用 Level3 且不视警告为错误编译
当主分支（v7 适配分支）也这样调整后，我会自动关闭 issue，希望理解

closed 2021-04-23 by MouriNaruto