lvgl/lv_port_pc_visual_studio #28: For using the the lvgl (v8) examples

ataweg, 2021-05-01:
please add 
`#include "examples/widgets/lv_example_widgets.h"`
to LVGL_Simulator.cpp and you can use the lvgl examples.

MouriNaruto, 2021-05-01:
Thanks. I want to borrow something from your link.
I will test some examples you said which have issues. (And upgrade LVGL submodules for trying to solve the issue.)

kisvegabor, 2021-05-02:
@ataweg 
I've fixed the mentioned examples.

MouriNaruto, 2021-05-05:
The mentioned examples seem to be fixed after I upgrade the LVGL submodules.
https://github.com/lvgl/lv_sim_visual_studio/commit/faac4b025661e50d23629958b67ad73384ab1f0d.

MouriNaruto, 2021-06-04:
It looks the issue is solved. I think it's time to close the issue. If have any question, please reply to this issue.

closed 2021-06-04 by MouriNaruto