lvgl/lv_port_pc_visual_studio #29: visual studio for v8 Error

bevis-wong, 2021-06-03:
i loading lvgl_master in simulate and the following error occurred
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\um\winres.h(1): fatal error RC1011: compiler limit : 'C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\Include\um;': macro definition too big
Locate to the following file "winres.h", How to do that ??

MouriNaruto, 2021-06-03:
I have recompiled this repo after recloning it to the separate folder and it's OK.

Maybe you need to reclone the repo because the master branch of this repo is migrate to LVGL v8 and had done a lot of modifications.

bevis-wong, 2021-06-03:
I use the master right now, and the problem the same. my steps as follows:
1. clone lv_sim_visual_studio  master
2. loading the file as
            LVGL.Simulator/lv_demos ( new Tag )
            LVGL.Simulator/lv_drivers ( master )
           LVGL.Simulator/lvgl (master or  new Tag)
           LVGL.Simulator/lv_demo_conf.h (re-name & copy from lv_demos)
3. Run "LVGL.Simulator.sln"
4. Set Debug to X64 or X86
5. You can see my problem

MouriNaruto, 2021-06-03:
My clone command is:
> git clone --recurse-submodules https://github.com/lvgl/lv_sim_visual_studio

I hadn't changed the files, and open the "LVGL.Simulator.sln", compile all targets and can't see you issue because it's succeeded.

bevis-wong, 2021-06-03:
yes, I follow your steps and compile all target automatically. they still show that
"resource.h(1): fatal error RC1011: compiler limit : 'C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\Include\um;': macro definition too big".
Maybe the log show that the "resource.h" and relate to macro "IDI_LVGL" ???
```
    if (!lv_win32_init(
        GetModuleHandleW(NULL),
        SW_SHOW,
        800,
        480,
        LoadIconW(GetModuleHandleW(NULL), MAKEINTRESOURCE(IDI_LVGL))))
    {
        return -1;
    }
```

MouriNaruto, 2021-06-03:
@bevis-wong I hope you can upload the log which is BuildAllTargets.cmd returns.

MouriNaruto, 2021-06-03:
And the CI is OK too. (https://github.com/lvgl/lv_sim_visual_studio/runs/2717679305)

icodk, 2021-06-09:
Jus cloned the code using :
`git clone --recurse-submodules https://github.com/lvgl/lv_sim_visual_studio.git`
Tried to build (for both x86 and x64) and got the following error in the console:

```
1>------ Build started: Project: LVGL.Simulator, Configuration: Debug x64 ------
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v142 (Platform Toolset = 'v142') cannot be found. To build using the v142 build tools, please install v142 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution".
1>Done building project "LVGL.Simulator.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

And in the error list:

```
Error	MSB8020	The build tools for v142 (Platform Toolset = 'v142') cannot be found. To build using the v142 build tools, please install v142 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution".	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets	LVGL.Simulator	57	

```

MouriNaruto, 2021-06-09:
@icodk It looks that you are using Visual Studio 2017.

In https://github.com/lvgl/lv_sim_visual_studio#introduction, it says you should use Visual Studio 2019.

icodk, 2021-06-09:
You are right I used 2017
2019 can't be installed at the moment. Microsoft issue:
`
Please note that this is a know issue that has been reported yesterday, the solution for this issue worked for some time and then, still happening again. This is related to a server outage from our end. I'll check if there's any ticket already created for this and I'll send you the link of it for follow up
`
and 
`Since the error has been reported for many users, it should be today because is not possible to download VS.`

The ticket (my be of interest for others):
https://developercommunity.visualstudio.com/t/Error-installing-visual-studio-community/1445949?space=61&entry=myfeedback

icodk, 2021-06-10:
VS19 can be installed at the moment and the simuæator works fine

MouriNaruto, 2021-07-19:
It seems OK, feel free to open a new issue if met the same issue.

closed 2021-07-19 by MouriNaruto