lvgl/lv_port_pc_visual_studio #30: Can't compile with Custom Font.

LiveSparks, 2021-06-16:
Declaring a custom font with `LV_FONT_DECLARE(impact_67);` results in the following linker error:
```
1>Target Link:
1>  gui.obj : error LNK2001: unresolved external symbol "struct _lv_font_t const impact_67" (?impact_67@@3U_lv_font_t@@B)
1>    Hint on symbols that are defined and could potentially match:
1>      impact_67
```
The same code and files compile and run successfully on the microcontroller.

I assume that for some reason Visual Studio is not linking the font file but I don't know how to force the issue.
The font file is visible in the solution explorer.
I tried to include .h file with the same name as the font and declare an extern variable in there but that doesn't work either.

Using this visual Studio Simulator would speed up my workflow by a large margin. Any help with this would be appreciated.

MouriNaruto, 2021-06-17:
It looks like you defined the `impact_67` as the C symbol, so the linker told you:

> Hint on symbols that are defined and could potentially match: impact_67

We all know, `#define LV_FONT_DECLARE(font_name) extern const lv_font_t font_name;`.

So if you define the font in `.c` file and use it in `.cpp` file, you should use `extern "C" const lv_font_t font_name;` to define it.

LiveSparks, 2021-06-17:
> So if you define the font in `.c` file and use it in `.cpp` file, you should use `extern "C" const lv_font_t font_name;` to define it.

Thanks, @MouriNaruto, that solved it. Didn't have to do this for the microcontroller for some reason even though that's also running C++.

closed 2021-06-17 by LiveSparks

MouriNaruto, 2021-06-17:
> Didn't have to do this for the microcontroller for some reason even though that's also running C++.

Maybe the compiler for microcontroller intends to export C symbol as possible.

a1103688841, 2021-08-20:
谢谢 非常感谢