lvgl/lv_port_pc_visual_studio #31: Project is `not loaded`

sky5454, 2021-07-24:
[image]

MouriNaruto, 2021-07-24:
Which version you have used in the MSVC toolchain.

sky5454, 2021-07-24:
> Which version you have used in the MSVC toolchain?

VS2019 16.10.4
v142 Latest
[image]

MouriNaruto, 2021-07-24:
It looks interesting because I use latest v142 toolchain too.

Could you export the Visual Studio installer configuration? I wonder to reproduce the issue.

sky5454, 2021-07-25:
> It looks interesting because I use latest v142 toolchain too.
> 
> Could you export the Visual Studio installer configuration? I wonder to reproduce the issue.
> 
> Updated: You may need to install the Windows SDK in the Visual Studio installer.

```json
{
  "version": "1.0",
  "components": [
    "Microsoft.VisualStudio.Component.CoreEditor",
    "Microsoft.VisualStudio.Workload.CoreEditor",
    "Microsoft.NetCore.Component.Runtime.5.0",
    "Microsoft.NetCore.Component.Runtime.3.1",
    "Microsoft.NetCore.Component.SDK",
    "Microsoft.VisualStudio.Component.NuGet",
    "Microsoft.Net.Component.4.6.1.TargetingPack",
    "Microsoft.VisualStudio.Component.Roslyn.Compiler",
    "Microsoft.VisualStudio.Component.Roslyn.LanguageServices",
    "Microsoft.VisualStudio.Component.FSharp",
    "Microsoft.ComponentGroup.ClickOnce.Publish",
    "Microsoft.NetCore.Component.DevelopmentTools",
    "Microsoft.VisualStudio.ComponentGroup.WebToolsExtensions",
    "Microsoft.Net.Component.4.8.SDK",
    "Microsoft.Net.Component.4.7.2.TargetingPack",
    "Microsoft.Net.ComponentGroup.DevelopmentPrerequisites",
    "Microsoft.VisualStudio.Component.TypeScript.4.2",
    "Microsoft.VisualStudio.Component.JavaScript.TypeScript",
    "Microsoft.Component.MSBuild",
    "Microsoft.VisualStudio.Component.TextTemplating",
    "Microsoft.VisualStudio.Component.Common.Azure.Tools",
    "Microsoft.VisualStudio.Component.SQL.CLR",
    "Microsoft.VisualStudio.Component.ManagedDesktop.Core",
    "Microsoft.Net.Component.4.5.2.TargetingPack",
    "Microsoft.Net.Component.4.5.TargetingPack",
    "Microsoft.Net.Component.4.TargetingPack",
    "Microsoft.Net.Component.4.5.1.TargetingPack",
    "Microsoft.Net.Component.4.6.TargetingPack",
    "Microsoft.Net.ComponentGroup.TargetingPacks.Common",
    "Microsoft.VisualStudio.Component.DiagnosticTools",
    "Microsoft.VisualStudio.Component.EntityFramework",
    "Microsoft.VisualStudio.Component.Debugger.JustInTime",
    "Microsoft.VisualStudio.Component.IntelliCode",
    "Microsoft.Net.Core.Component.SDK.2.1",
    "Microsoft.VisualStudio.Component.VC.CoreIde",
    "Microsoft.VisualStudio.Component.VC.Tools.x86.x64",
    "Microsoft.VisualStudio.Component.Graphics.Tools",
    "Microsoft.VisualStudio.Component.VC.DiagnosticTools",
    "Microsoft.VisualStudio.Component.Windows10SDK.19041",
    "Microsoft.VisualStudio.Component.ManagedDesktop.Prerequisites",
    "Microsoft.ComponentGroup.Blend",
    "Microsoft.VisualStudio.Component.DotNetModelBuilder",
    "Microsoft.VisualStudio.Workload.ManagedDesktop",
    "Microsoft.VisualStudio.Component.VC.Redist.14.Latest",
    "Microsoft.VisualStudio.ComponentGroup.NativeDesktop.Core",
    "Microsoft.VisualStudio.ComponentGroup.WebToolsExtensions.CMake",
    "Microsoft.VisualStudio.Component.VC.CMake.Project",
    "Microsoft.VisualStudio.Component.VC.ATL",
    "Microsoft.VisualStudio.Component.VC.TestAdapterForBoostTest",
    "Microsoft.VisualStudio.Component.VC.TestAdapterForGoogleTest",
    "Microsoft.VisualStudio.Component.VC.CLI.Support",
    "Microsoft.VisualStudio.Component.VC.ASAN",
    "Microsoft.Component.VC.Runtime.UCRTSDK",
    "Microsoft.VisualStudio.Workload.NativeDesktop"
  ]
}
```

and all is the lastest.

MouriNaruto, 2021-07-27:
@sky5454 

I have tried the configuration you exported in my test VM, and I found everything is OK.

I want to suggest you to reinstall the Windows 10 SDK when I see you screenshot of Visual Studio Installer.

MouriNaruto, 2021-08-03:
It seems the issue had been solved. I think it's time to close the issue. If you have the same issue, please fire a new issue.

closed 2021-08-03 by MouriNaruto