lvgl/lv_port_pc_visual_studio #32: Install problem

geoffw123, 2021-08-10:
Hi

I used the suggested git clone --recurse-submodules https://github.com/lvgl/lv_sim_visual_studio.git command to install this project on Windows 10.

I am assuming (maybe wrongly) that this should download all the source code needed for this project ?
I have got some code downloaded but for example folders D:\lv_sim_visual_studio\LVGL.Simulator\lvgl, D:\lv_sim_visual_studio\LVGL.Simulator\lv_demos and others are present but totally empty ?

Any help on fixing this please ?  Thanks

MouriNaruto, 2021-08-10:
I have do that in my test environment. It seems OK.

So I think you may meet the network connection issue when you clone the repo. So may be you need to reclone again.

geoffw123, 2021-08-10:
hiya thanks for the speedy reply. I have done it 3 times, same empty folders every time.

embeddedt, 2021-08-10:
What is the output from Git on the terminal?

geoffw123, 2021-08-10:
when I tried it this morning the git console command finished very quickly. However it gave no error messages at all (Tried this 3 times, same every time)
Just tried it again this afternoon, it took 3 times longer and downloaded much more stuff, the folders now have content in them.
Have you changed anything your end ?  Otherwise must have been a glitch with github.
Thanks for your help

geoffw123, 2021-08-10:
Yeah must have been a Github problem. I have got it building and running now, neat demo App. :)
Why is the archive config set for ARM on a Windows Simulator ? That threw me for a little while til I noticed it and changed it to x86.