lvgl/lv_port_pc_visual_studio #33: Download problems again

geoffw123, 2021-08-24:
I eventually managed to download the full simulator archive couple of weeks ago and got it working. However today it blew up with some memory pool error that I didnt understand, so I thought I would just delete and download again and start from scratch. However I have got the partial zip file problem back, I been trying to download the V8 master zip all day. It always downloads a zip but it has several folders with nothing in lvgl folder for example. I tried 3 ways to get the source code all failed.  Please fix this problem or put the correct zip file somewhere else so I can download it please

1) tried just clicking on the "download zip" here. Failed as above

2) tried git clone https://github.com/lvgl/lv_sim_visual_studio from cmd prompt. 
   It worked and downloaded files, but once again folders are empty

3) tried alternative source site from a google search.  
   https://lvgl.io/pc-simulator gives a 404 not found

Thanks

embeddedt, 2021-08-24:
You need `git clone --recursive https://github.com/lvgl/lv_sim_visual_studio`, otherwise the submodules won't be cloned.

geoffw123, 2021-08-25:
ah thanks for the cmd line correction, that fixed the git download for me as expected (I dont normally use cmd line for this so missed that)

Just for info. I again tested my usual method of clicking the "Download Zip" button. Same result again this morning, all folders in the zip below the 2nd level are empty still. No idea why

Least the archive version is now running again and has fixed my memory heap walker crash. I can start experimenting again to see what caused the crash for me.

embeddedt, 2021-08-25:
> Just for info. I again tested my usual method of clicking the "Download Zip" button. Same result again this morning, all folders in the zip below the 2nd level are empty still. No idea why

Unfortunately, we have no control over that, as GitHub still (!) doesn't include submodules in the ZIP they generate. It would be nice to just hide the ZIP link but unfortunately we don't get that option either. :laughing:

closed 2022-04-27 by stale[bot]