lvgl/lv_port_pc_visual_studio #34: Small bug with Slider control ?

geoffw123, 2021-08-29:
Hi
I am guessing this is a V8 simulator bug with the slider ?  The light blue semi circles at the end of the slider background are not drawing properly ?  I havent tried this with V8 on my hardware, so cant say if its specific to simulator or not. See attached screenshot.
[image]

kisvegabor, 2021-08-30:
It was a bug in lvgl master a few weeks ago. Updating lvgl should solve the issue.

MouriNaruto, 2021-09-01:
I have update all submodules in this repo. I hope this issue can be solved.

geoffw123, 2021-09-02:
Thanks for the speedy update. Confirmed fixed for me

MouriNaruto, 2021-09-02:
It seems to be OK to close this issue.

closed 2021-09-02 by MouriNaruto