lvgl/lv_port_pc_visual_studio #35: Branch release/v7, vs running problem

gregrgr, 2021-09-03:
Use the following command clone 
git clone -b release/v7 --recurse-submodules https://github.com/lvgl/lv_sim_visual_studio.git
[image]

No response when running vs.

sergioaiello, 2022-01-07:
Hi.
I'm having the same problem.
I need to install v7 on visual studio 2022 (actually downgraded 2019 v142) and I do not find the way.
Please help!
thank you

MouriNaruto, 2022-01-12:
@gregrgr @sergioaiello 

I have reproduced the issue and fixed it.

Because the newer version of win32drv don't support LVGL v7, I have created the win32drv_v7 in release/v7 branch.

Kenji Mouri

closed 2022-01-12 by MouriNaruto

sergioaiello, 2022-01-12:
Hi,
I did:
git clone https://github.com/lvgl/lv_sim_visual_studio.git
--recurse-submodules
cd lv_sim_visual_studio
git checkout release/v7
git submodule update

when I loaded the sln file I got an error window with the following
message:
---------------------------
One or more projects in the solution were not loaded correctly.
Please see the Output Window for details.
-------------------------

And Visual Studio did not open a workable project.

Could you, pls, help me?

Thank you and best regards, S.