lvgl/lv_port_pc_visual_studio #45: Enable file system on simulator

jacquesdmernok, 2022-08-20:
Hi there. Has anyone been able to successfully add the file system support to the simulator? 

I would like to use the simulator to create demos of user interactions. But I plan to have quite a bit of images and gif files. It would be great if I could have a directory on the PC with these files, and reference the files from within the simulator.

kisvegabor, 2022-08-24:
You can simply use `LV_USE_FS_WIN32` in `lv_conf.h`.

MouriNaruto, 2022-08-24:
Note: You can get the commented example in `LVGL.Simulator.cpp` itself.

https://github.com/lvgl/lv_port_win_visual_studio/blob/29e7abe13c1c5ed21f9d7e104628fa0afa83ba91/LVGL.Simulator/LVGL.Simulator.cpp#L88-L101

Kenji Mouri

MouriNaruto, 2022-12-28:
Close issue because this issue should be solved.

Kenji Mouri

closed 2022-12-28 by MouriNaruto