lvgl/lv_port_pc_visual_studio #47: What happened?

break2133, 2022-12-23:
[image]

kisvegabor, 2022-12-23:
Thank you for pointing it out. I'm sure it was left there by accident. 

It was added in https://github.com/lvgl/lv_port_win_visual_studio/commit/5a44f6408b29751461e4efb266b77babb0c045cb.  @MouriNaruto, could you comment on it?

I've just removed it in https://github.com/lvgl/lv_port_win_visual_studio/commit/75f72844ef0aa9921013c60cb5f83236f7bfef89.

MouriNaruto, 2022-12-23:
@break2133 @kisvegabor 

> could you comment on it?

I am used to write a “fuck” function with no parameters and "void" return type first before modifying to the definition of new function definition needed because I couldn't think of the correct definition of the function in one breath beforehand. It's a scaffold-like thing which is left accidentally when I am implementing the example of multiple monitor support of win32drv.

I am sorry about that. Thanks @kisvegabor for removing that.

Kenji Mouri

MouriNaruto, 2022-12-28:
Close issue because this issue should be solved.

Kenji Mouri

closed 2022-12-28 by MouriNaruto

kisvegabor, 2023-01-04:
Sorry for the late reply, I'm still catching up with the emails after the Christmas holiday.

> I will try to use "void test()" instead of "void fuck()" before I modify to the definition of new function definition needed for reducing the similar thing happened. (The similar thing had also been happened in the project which I created, here is the example: https://github.com/M2Team/NanaZip/commit/7149e36809748b4fba40e6ed98a1f335df61b108)

It think it'd be safer, thank you :relaxed: