lvgl/lv_port_pc_visual_studio #50: Text render error with chinese char.

Coloryr, 2023-01-03:
```c
lv_obj_t* obj = lv_obj_create(lv_scr_act());
    lv_obj_set_align(obj, LV_ALIGN_CENTER);
    lv_obj_set_width(obj, 500);
    lv_obj_set_height(obj, 300);

    lv_obj_t* obj1 = lv_obj_create(obj);
    lv_obj_align(obj1, LV_ALIGN_TOP_LEFT, 5, 5);
    lv_obj_set_size(obj1, 120, 70);
    lv_obj_t* label = lv_label_create(obj1);
    lv_obj_set_style_text_font(label, &lv_font_simsun_16_cjk, 0);
    lv_label_set_text(label, "测试");
    lv_obj_center(label);

    button1 = lv_btn_create(obj);
    lv_obj_add_event_cb(button1, event_handler, LV_EVENT_ALL, NULL);
    lv_obj_align(button1, LV_ALIGN_TOP_LEFT, 5, 85);
    lv_obj_set_size(button1, 120, 70);
    label = lv_label_create(button1);
    lv_obj_set_style_text_font(label, &lv_font_simsun_16_cjk, 0);
    lv_label_set_text(label, "测试");
    lv_obj_center(label);
```

[image]

MouriNaruto, 2023-01-03:
You need to use lv_freetype to create a LVGL font object with simulator built-in FreeType.

I think I will port Windows GDI font backend from lv_port_windows to win32drv which used in simulator in recent days.

Kenji Mouri

Coloryr, 2023-01-04:
use `lv_freetype` or `make one font by myself` was work well.

closed 2023-01-04 by Coloryr