lvgl/lv_port_pc_visual_studio #52: Request for a a release/v8.3 branch

kisvegabor, 2023-01-05:
Hi,

One of our customer is using this simulator but in their project we are using LVGL v8.3.

They could manage to switch to LVGL v8.3 from the master branch but it'd be great to have a dedicate branch for it. 

What do you think? If you agree could you add it?

MouriNaruto, 2023-01-05:
OK, I will create a release/v8.3 branch and switch master to 9.0-dev as soon as possible.

Kenji Mouri

kisvegabor, 2023-01-05:
Thank you! :relaxed:

MouriNaruto, 2023-01-06:
The "release/v8.3" branch is ready. So, I closed the issue.

https://github.com/lvgl/lv_port_win_visual_studio/tree/release/v8.3

Kenji Mouri

closed 2023-01-06 by MouriNaruto

kisvegabor, 2023-01-06:
Thank you very much. They've already reported that it's working well :slightly_smiling_face: