lvgl/lv_port_pc_visual_studio #53: [Roadmap] Big refactoring of this repository

MouriNaruto, 2023-01-09:
In order to better maintain LVGL for Windows Visual Studio port, I have planned to do a big refactoring for this repository. I think it may be the biggest refactoring since I become the maintainer of this repository. Here is the list of things I need to do.

- [ ] Merge the features from [lv_port_windows](https://github.com/lvgl/lv_port_windows) for improving the maintain work and simplify the LVGL users choice.
  - [ ]  Create a project called LvglWindowsDesktopApplication project in this repository for migrating the LVGL.Windows.Desktop project from [lv_port_windows](https://github.com/lvgl/lv_port_windows).
  - [ ] Add several features for [win32drv](https://github.com/lvgl/lv_drivers/tree/master/win32drv).
    - [ ] Add the "Application Mode" for using LVGL to develop Windows desktop applications.
    - [ ] Add window resizing support for "Application Mode".
    - [ ] Add Per Monitor DPI Awareness V1 (Windows 10, but before Build 14986) and V2 (Windows 10 Build 14986 and later) support for "Application Mode".
    - [ ] Add Input Method Editors (IME) support for "Application Mode".
  - [ ] Add Windows GDI font engine backend to [LVGL main repository](https://github.com/lvgl/lvgl)
  - [ ] Update the ReadMe document of [lv_port_windows](https://github.com/lvgl/lv_port_windows) for noticing user the repository will be archived and notify @kisvegabor.
- [ ] Tidy the implementation of [LVGL.Simulator](https://github.com/lvgl/lv_port_win_visual_studio/tree/master/LVGL.Simulator).
  - [ ] Rename LVGL.Simulator to LvglWindowsSimulator for resolving https://github.com/lvgl/lv_port_win_visual_studio/issues/42.
  - [ ] Create MSBuild shared item project for hosting the submodules used in this repository for resolving https://github.com/lvgl/lv_port_win_visual_studio/issues/49.
 - [ ] Improve the performance and fix bugs for [win32drv](https://github.com/lvgl/lv_drivers/tree/master/win32drv).
 - [ ] Implement the win32drv vNext with the [new driver architecture](https://github.com/lvgl/lvgl/tree/arch/drivers)

Kenji Mouri

kisvegabor, 2023-01-10:
Sounds like a good plan!

MouriNaruto, 2023-05-17:
Finally, I have implemented the MSBuild project updater properly in https://github.com/lvgl/lv_port_win_visual_studio/blob/master/LvglMaintainerTools/LvglSubmoduleProjectFileGenerator/Program.cs. Users can change the files in lvgl and freetype filter in Visual Studio IDE freely and resolve the issue https://github.com/lvgl/lv_port_win_visual_studio/issues/49.

I can finally to push the progress for merging the features from [lv_port_windows](https://github.com/lvgl/lv_port_windows) for improving the maintain work and simplify the LVGL users choice.

Kenji Mouri

kisvegabor, 2023-05-18:
> Users can change the files in lvgl and freetype filter in Visual Studio IDE freely and resolve the issue https://github.com/lvgl/lv_port_win_visual_studio/issues/49.

It's amazing, it was an issue for some of our customers a couple of times. 

> I can finally to push the progress for merging the features from [lv_port_windows](https://github.com/lvgl/lv_port_windows) for improving the maintain work and simplify the LVGL users choice.

It's also great!  Does it mean some parts can be added to LVGL master too? Or is it an other story?

MouriNaruto, 2023-05-18:
@kisvegabor 

> Does it mean some parts can be added to LVGL master too? 

In sort of, yes. I want to provide a better implementation for LVGL, so I tangled quite a few things. It makes me reduce the something which I tangled.

For example, I want to provide a flexible and modular implementation for people, make people can integrate LVGL content (simulator mode or application mode) in other windows easily. I think it will make LVGL users happy because it can bring many possibilities. This is why I am tangled for implementing the new Windows backend. (I want to create a better design.)

Here are the things I want to provide in new Windows backend.

- LVGL content hosting support in windows which provides the HWND for interop with other UI frameworks like MFC, Qt, WinForms, WPF etc. (Maybe we can use that to implement the LVGL Designer for Visual Studio.)
- Multiple render backend support. In the current stage, I think I will only provide the Windows GDI backend. (But I hope I can provide the Direct2D backend in the future for reducing the overhead when Desktop Window Manager to composite the Window contents and have a good scaling quality for the simulator mode.)
- I also hope I can implement UI Automation support. It can make LVGL supports accessibility in Windows Vista and later. I have several friends who are blind, this is why I care about accessibly. Also, users can use that to do unit tests for LVGL content in Windows.

Kenji Mouri

kisvegabor, 2023-05-19:
Sounds great! 👍

From the GitHub star count clearly the Windows port repos are the most popular ones, so I'm really happy that they will be even better 😊

MouriNaruto, 2023-05-23:
@kisvegabor 

I hope this repository could be renamed to lv_port_pc_visual_studio.

Because one of my friends created the project which can build Linux projects managed by Visual Studio project file in the local Linux instance. Here is the repository: https://github.com/Chuyu-Team/MSBuildCppCrossToolset

[image]

I will utilize this project to create a Visual Studio project which is for LVGL Linux Wayland application demo for showing more possibilities for users.

Kenji Mouri

kisvegabor, 2023-05-23:
Woooow, it sounds amazing! Just let me when can I rename it :blush:

MouriNaruto, 2023-05-23:
@kisvegabor 

> Woooow, it sounds amazing! Just let me when can I rename it 😊

It's time to rename now and I will commit these changes recently.

Kenji Mouri

kisvegabor, 2023-05-23:
Done :slightly_smiling_face:

MouriNaruto, 2023-05-23:
@kisvegabor 

Also, the https://github.com/lvgl/lv_port_windows repository can be archived now because I have updated the deprecated information.

I am working on the replacement implementation now and I can commit them to this repository today or tomorrow.

Kenji Mouri

MouriNaruto, 2024-01-15:
Except for adding Windows GDI font engine backend. All related work has done by https://github.com/lvgl/lvgl/pull/5313.

Kenji Mouri

closed 2024-01-15 by MouriNaruto