lvgl/lv_port_pc_visual_studio #54: about compile fail

charles-lyc, 2023-06-03:
Is win sim ready?  I pull the release/v8.3 branch, which is still not compiled success.

MouriNaruto, 2023-06-03:
No big changes in the release/v8.3 branch. So, it should be successful if you clone them properly with the git command line.

Kenji Mouri

charles-lyc, 2023-06-04:
I pull

charles-lyc, 2023-06-04:
I pull like this:
[image]

charles-lyc, 2023-06-04:
And seems the lv_drivers repo didn't work, before I try master, still mail function.
[image]

MouriNaruto, 2023-06-04:
Please clone release/v8.3 branch directly via git clone command.

Because repository structure of master (9.x) branch is huge different.

Kenji Mouri

charles-lyc, 2023-06-04:
Sure, but it failed in some reason:
[image]
So I pull them one by one.

charles-lyc, 2023-06-04:
Seems like the commit ID was not found? I'm not sure...

charles-lyc, 2023-06-04:
It works!

charles-lyc, 2023-06-04:
It seems like, I pull the code from a mirror website, but it didn't contain some specific commit.
Thx.