lvgl/lv_port_pc_visual_studio #55: Multi-window support

kisvegabor, 2023-08-13:
Hi,

I wonder if it's possible to create 2 windows for 2 `lv_disp_t` instances with Visual Studio.

It would very useful for a particular project where we have the normal UI in one window and we add some slider to simulate data changes in an other window.

MouriNaruto, 2023-08-14:
Yes, it's possible to create 2 windows.

Kenji Mouri

kisvegabor, 2023-08-14:
Great! Could you shown an example for that? Or shall we just call `lv_win32_init` again?

MouriNaruto, 2023-08-14:
I will add the example for that as soon as possible. Please wait for a while. If it's OK, I will update the progress in this issue.

Kenji Mouri

kisvegabor, 2023-08-14:
Thank you, Kenji!

MouriNaruto, 2024-10-09:
The multi window support actually added in LVGL 9.x's Windows backend.

Kenji Mouri

closed 2024-10-09 by MouriNaruto