lvgl/lv_port_pc_visual_studio #56: Build errors with LVGL v8.3.9

zjanosy, 2023-08-15:
I have checked out the release/v8.3 branch, and I have updated the lvgl submodule to the latest v8.3.9 version. The project worked with the original v8.3.5 version. Now the Visual Studio project does not compile with the errors:
```
1>c1 : fatal  error C1083: Cannot open source file: 'lvgl\src\draw\nxp\vglite\lv_gpu_nxp_vglite.c': No such file or directory
1>c1 : fatal  error C1083: Cannot open source file: 'lvgl\src\draw\nxp\lv_gpu_nxp.c': No such file or directory
```
Apparently these files have been removed from the repo, but they are in the VS project. I tried removing the complete nxp folder manually from the project, but got an error message:
```
Cannot modify an evaluated object originating in an imported file "D:\lvgl\test\lv_port_pc_visual_studio_v8\LVGL.Simulator\LVGL.Portable.vcxitems".
```
I use Visual Studio 2022 Community Edition.

MouriNaruto, 2023-08-15:
I will update it for fixing the issue.

Kenji Mouri

MouriNaruto, 2023-08-16:
The issue in release/v8.3 branch should be fixed.

Kenji Mouri

zjanosy, 2023-08-16:
Thank you! It is working now.

closed 2023-08-16 by zjanosy

kisvegabor, 2023-08-16:
Thank you for the prompt update, Kenji!