lvgl/lv_port_pc_visual_studio #57: Not working with latest master

kisvegabor, 2023-08-17:
Hi Kenji,

I've noticed that the latest changes of LVGL `master` are not followed in the repository yet and the build failes. Most of the conceptual changes are already added so we don't expect large and complex changes in the near future. 

Do you plan to take look at it at some time?

FYI, here are the changes of v9 so far: 
- https://github.com/lvgl/lvgl/issues/4011
- https://github.com/lvgl/lvgl/issues/4016#issue-1599634031

MouriNaruto, 2023-08-18:
Of course, I synchronized the LVGL submodule yesterday.

I am adapting the Win32 backend to the newest LVGL. ("Most of the conceptual changes are already added so we don't expect large and complex changes in the near future." is really nice for me.)

Kenji Mouri

kisvegabor, 2023-08-18:
Splendid, thank you!

kisvegabor, 2023-08-23:
> Also, I will add parallel rendering adaption for Windows in recent days.

Could send an update with a  just working project and make the parallel rendering update as a separate step? It's just to make simulator work with latest master as soon as possible.

MouriNaruto, 2023-08-23:
@kisvegabor 

> Could send an update with a just working project and make the parallel rendering update as a separate step? It's just to make simulator work with latest master as soon as possible.

I have fixed the build failures just now.

https://github.com/lvgl/lv_port_pc_visual_studio/actions/runs/5952165857/job/16143522062

kisvegabor, 2023-08-25:
Awesome, thank you Kenji.

@zjanosy could you try it out?