lvgl/lv_port_pc_visual_studio #60: Render Issue

xland, 2023-12-12:
Windows 10 19044.2846

lv_port_pc_visual_studio main branch

[image]

In addition to this

When I drag the mouse to change the window size, it causes the application crash.

MouriNaruto, 2023-12-12:
The backend in desktop application demo is deprecated in the future.

The new backend implementation in the simulator demo and it supports all features in desktop application demo.

If the new backend is finished, I will push them to the LVGL mainline.

Kenji Mouri

MouriNaruto, 2024-01-15:
The new backend is merged in the LVGL mainline.

https://github.com/lvgl/lvgl/pull/5313

Kenji Mouri

closed 2024-01-15 by MouriNaruto