lvgl/lv_port_pc_visual_studio #63: Size of the screen is not perfect.

Kirbyrawr, 2023-12-31:
# Problem
When you specify 800x480 for example, the content is less than that since is taking space from the title bar.

#Potential Solution
When initializing the screen in the wind32drv.c, we can modify the CreateWindowExW in this way:

```
HWND display_window_handle = CreateWindowExW(
    NULL,
    LVGL_SIMULATOR_WINDOW_CLASS,
    window_title,
    WS_POPUP,
    CW_USEDEFAULT,
    0,
    hor_res,
    ver_res,
    NULL,
    NULL,
    instance_handle,
    NULL);
```

Basically we change 
- WINDOW_EX_STYLE to NULL
- WINDOW_STYLE to WS_POPUP

This way the screen wont have title bar but the content size will be the exact one.
[image]

MouriNaruto, 2024-01-02:
> When you specify 800x480 for example, the size of the content screen is less than that since is taking space from the title bar.

If you have read the implementation, you will know the implementation will adjust to the correct content resolution. So, it won’t affect.

Kenji Mouri

Kirbyrawr, 2024-01-03:
> If you have read the implementation, you will know the implementation will adjust to the correct content resolution. So, it won’t affect. [...]

Hi Kenji, that looks like it should do the trick but then why if you take a screenshot of it, it doesn't have the same pixel size?

Kirbyrawr, 2024-01-21:
Hi, regarding this problem, it looks like it's related to the DPI of 125% from Windows of my screen, 400 x 300 is 320 x 240 * 1.25 that is my scale.
I tried to put the overrideDPI as false and true and nothing changes there.

Kirbyrawr, 2024-01-21:
Even in the latest commit is still like that, i'm not sure if i'm doing something wrong.

MouriNaruto, 2024-01-22:
> Hi, regarding this problem, it looks like it's related to the DPI of 125% from Windows of my screen, 400 x 300 is 320 x 240 * 1.25 that is my scale.

https://docs.lvgl.io/master/integration/driver/windows.html

Maybe this doc will do some help.

Kenji Mouri

Kirbyrawr, 2024-01-22:
Hi there @MouriNaruto thanks for the link, sadly it doesn't work. Let me explain the situation...

If i run LVGLSimulator project with DPI 100% in Windows, it works. ✅ 
If i run LVGLSimulator project with DPI 125% in Windows, it doesn't work. ❌ 

I tried both simulator and desktop mode, also i tried to change DPI Override etc..., it doesn't work.

Kirbyrawr, 2024-01-22:
DPI 100% Simulator Mode:
[image]

DPI 125% Simulator Mode:
[image]

You can check that the size is different, changing the desktop and dpi in code doesn't work.

MouriNaruto, 2024-01-22:
Got it. I will try to do some fix.

Kenji Mouri

19e2k67, 2024-06-29:
[image]
my computer DPI = 125%, I change `LvglWindowsSimulator.cpp` 26 line
```c
bool simulator_mode = false;
```
it present 100%
[image]
Default is true, display 125%, font is blurry
[image]