lvgl/lv_port_pc_visual_studio #65: when i update the lvgl to the newest version, so error!!

xinsuinizhuan, 2024-01-12:
1、lv_indev_get_disp not define
error as:
win32drv.obj : error LNK2019: 无法解析的外部符号 lv_indev_get_disp，函数 lv_windows_get_indev_window_handle 中引用了该符号
2、not show ui, tips:
 _lv_display_refr_timer: No draw buffer lv_refr.c:350

xinsuinizhuan, 2024-01-12:
new verison has no lv_indev_get_disp,but in win32drv.chas the function:
[image]

MouriNaruto, 2024-01-12:
Because LVGL's new main branch changed some ABI. I need to follow. Please wait.

Kenji Mouri

MouriNaruto, 2024-01-12:
https://github.com/lvgl/lvgl/commit/e10f573150a95822df6422541bcc9ea1238f098a#diff-35a05ebbf5595318ed115bcabc7f70e6ae0b4f1f3ace3629e154a7434fe680f1

It looks the yesterday's LVGL mainline changed a lot of API namings.

Kenji Mouri

MouriNaruto, 2024-01-12:
https://github.com/lvgl/lv_port_pc_visual_studio/commit/aa94c0504b8e1a1836d424723c46044cba5f5ee6

It should be fixed after this commit.

Kenji Mouri

xinsuinizhuan, 2024-01-12:
still error:
when i update the newest version, open the .sln，then errors:
c1 : fatal error C1083: coould not open: “E:\lvgl\git\lv_port_pc_visual_studio-master\LvglPlatform\lvgl\src\draw\renesas\dave2d\lv_draw_dave2d_uitls.c”: No such file or directory

xinsuinizhuan, 2024-01-12:
still:
[image]

xinsuinizhuan, 2024-01-12:
still:
[image]

MouriNaruto, 2024-01-12:
I don't think you have updated the newest version of this repository.

Because this old issue I have fixed.

Kenji Mouri

MouriNaruto, 2024-01-12:
[image]

With the GitHub search feature, you will know I fixed this issue.

Kenji Mouri

MouriNaruto, 2024-10-09:
Close this issue due to no further replies.

Kenji Mouri

closed 2024-10-09 by MouriNaruto