lvgl/lv_port_pc_visual_studio #67: use the defalut lv_conf_template.h change to lv_conf.h， _lv_display_refr_timer: No draw buffer lv_refr.c:349

xinsuinizhuan, 2024-01-22:
I change the defalut lv_conf_template.h change to lv_conf.h，and open some define，then break:
[Warn]  (9592.984, +9592984)     lv_demo_widgets: LV_FONT_MONTSERRAT_16 is not enabled for the widgets demo. Using LV_FONT_DEFAULT instead. lv_demo_widgets.c:131
[Warn]  (9592.984, +0)   _lv_display_refr_timer: No draw buffer lv_refr.c:349

E:\lvgl\git\lv_port_pc_visual_studio-master\Output\Binaries\Release\x64\LvglWindowsSimulator.exe

xinsuinizhuan, 2024-01-22:
lv_conf.h file：
[lv_conf_old.zip](https://github.com/lvgl/lv_port_pc_visual_studio/files/14004333/lv_conf_old.zip)

kisvegabor, 2024-01-22:
It should be already fixed. Be sure to use the latest lv_port_pc_visual_studio and lvgl version.

kisvegabor, 2024-01-22:
cc @MouriNaruto

MouriNaruto, 2024-01-22:
I will investigate it.

Kenji Mouri

Gerken80, 2024-06-13:
_lv_display_refr_timer: No draw buffer lv_refr.c:349

I am getting this issue. Is it a config issue?