lvgl/lv_port_pc_visual_studio #68: 可以在windows应用中增加一个缩小到系统托盘的功能吗

jinsc123654, 2024-03-13:
我想使用lvgl做一个windows小工具，它需要在后台运行，但是lvgl缩小后是在下方导航栏
我需要把lvgl的应用收进系统托盘栏，请问我需要怎么去修改对应源码
类型下方的缩小进系统导航栏
[image]

kisvegabor, 2024-03-14:
Please write in English. Using a translator is fine too.

jinsc123654, 2024-03-14:
Chinese translation:
I want to use lvgl to create a Windows mini tool that needs to run in the background, but when lvgl is scaled down, it will appear in the navigation bar below
I need to put the lvgl application into the system tray. May I know how to modify the corresponding source code
Zoom out to the system navigation bar below the type

I have already implemented this function myself, thank you
[image]

closed 2024-03-14 by jinsc123654