lvgl/lv_port_pc_visual_studio #69: some error

xinsuinizhuan, 2024-03-21:
one:
the windows application,max frame, the meter pointer is offline:
[image]

two:
the newest lvgl code， lv_image_header_cache_drop function is missing

MouriNaruto, 2024-03-21:
For the meter pointer, you need to ask for the widget demo in LVGL for resolution change support.

For the crash issue, I will try to solve.

Kenji Mouri

kisvegabor, 2024-03-22:
> For the meter pointer, you need to ask for the widget demo in LVGL for resolution change support.

Yeah, it's not fully responsive. :slightly_frowning_face:

MouriNaruto, 2024-10-09:
Close this issue due to no further replies.

Kenji Mouri

closed 2024-10-09 by MouriNaruto