lvgl/lv_port_pc_visual_studio #70: Update to the latest LVGL code

kisvegabor, 2024-04-29:
Hi,

With a customer we are trying two things:
1. Update LVGL from the master branch
2. Add a new folder with the UI code

We can't see how to add new folder or update the content of `lvgl` folder. Could you advise on that?

MouriNaruto, 2024-04-30:
I will update that and add a document for how to update lvgl folder manually.

Kenji Mouri

MouriNaruto, 2024-04-30:
I have updated the LVGL submodule to the latest.

Also, I have provided a document to synchronize VS solution manually.
https://github.com/lvgl/lv_port_pc_visual_studio/blob/master/Documents/HowToSynchronizeLvglRelatedSubmodules.md

Kenji Mouri

kisvegabor, 2024-05-06:
Thank you Kenji! 

Could you add a similar guide about adding/updating any custom submodules? (The typical workflow for us is to store the UI in a separate submodule.)