lvgl/lv_port_pc_visual_studio #71: The window is not responding in the Windows system

juxiangwu, 2024-05-04:
The following problems occur when LvglWindowsDesktopApplication is running: 
1. When the left mouse button is pressed and the layout is dragged, the whole window is unresponsive; 
2. After the left mouse button is released, move the mouse, the whole layout is also dragged, and after the left mouse button is clicked again, the whole window is unresponsive.

IDE:Visual Studio 2022
OS: Windows 11 x64
LVGL:9.1.1 dev

[image]
[image]
[image]

MouriNaruto, 2024-05-04:
I need to do some further investigation for that.

Kenji Mouri

648170003, 2024-05-07:
同样遇到此问题

648170003, 2024-05-07:
Encountering the same problem ， the problem still exists when i migrated from VS2019 to VS2022.