lvgl/lv_port_pc_visual_studio #72: black window after un-minimizing

sdicker8, 2024-05-11:
2022 build tools
win 11 x64
v9.1.1

When LvglWindowDesktopApplication is minimized and then un-minimized the window is black. When the app is then maximized the window recovers.

I also see #71 -- it seems to consume 100% of one of my cpu cores - the memory seems stable.

Thank you -
Scott

Nice project!

MouriNaruto, 2024-05-11:
I need to do some reproduce work to fix the issue.

Kenji Mouri

VogelPapaFinn, 2024-06-07:
Found a solution?

nikthefix, 2024-11-24:
The window needs repainting in full when un-minimizing otherwise only dirty areas will be painted when called.
Maybe use WM_WINDOWPOSCHANGED or similar.

shiweiping, 2024-12-16:
Found a solution?

ou2356, 2025-01-17:
Is there any remedy currently available?

ou2356, 2025-01-17:
> The window needs repainting in full when un-minimizing otherwise only dirty areas will be painted when called. Maybe use WM_WINDOWPOSCHANGED or similar.

Can you explain in detail how to remedy this problem? Thank you very much!

nikthefix, 2025-01-17:
@ou2356 I'm working on this at the moment. Are you using v9.0.0 release?

ou2356, 2025-01-17:
> [@ou2356](https://github.com/ou2356) I'm working on this at the moment. Are you using v9.0.0 release?

Yes, I am using v9.0.0 release. I am looking forward to hearing good news from you very much.

nikthefix, 2025-01-17:
This is working for me:

In lvglWindowsDesktopApplication/lvgl/src/drivers/windows/lv_windows_context.c

Insert:

 case WM_WINDOWPOSCHANGED: {} break;

In this context:

//////////////////////////////////////////////////////////////////////////////////////////

static bool lv_windows_window_message_callback_nolock(
    HWND hWnd,
    UINT uMsg,
    WPARAM wParam,
    LPARAM lParam,
    LRESULT * plResult)
{
    switch(uMsg) {
        case WM_CREATE: {
                // Note: Return -1 directly because WM_DESTROY message will be sent
                // when destroy the window automatically. We free the resource when
                // processing the WM_DESTROY message of this window.

                lv_windows_create_display_data_t * data =
                    (lv_windows_create_display_data_t *)(
                        ((LPCREATESTRUCTW)(lParam))->lpCreateParams);
                if(!data) {
                    return -1;
                }

                lv_windows_window_context_t * context =
                    (lv_windows_window_context_t *)(HeapAlloc(
                                                        GetProcessHeap(),
                                                        HEAP_ZERO_MEMORY,
                                                        sizeof(lv_windows_window_context_t)));
                if(!context) {
                    return -1;
                }

                if(!SetPropW(hWnd, L"LVGL.Window.Context", (HANDLE)(context))) {
                    return -1;
                }

                context->window_dpi = lv_windows_get_dpi_for_window(hWnd);
                context->zoom_level = data->zoom_level;
                context->allow_dpi_override = data->allow_dpi_override;
                context->simulator_mode = data->simulator_mode;

                context->display_timer_object = lv_timer_create(
                                                    lv_windows_display_timer_callback,
                                                    LV_DEF_REFR_PERIOD,
                                                    context);

                context->display_resolution_changed = false;
                context->requested_display_resolution.x = 0;
                context->requested_display_resolution.y = 0;

                context->display_device_object = lv_display_create(0, 0);
                if(!context->display_device_object) {
                    return -1;
                }
                RECT request_content_size;
                GetWindowRect(hWnd, &request_content_size);
                lv_display_set_resolution(
                    context->display_device_object,
                    request_content_size.right - request_content_size.left,
                    request_content_size.bottom - request_content_size.top);
                lv_display_set_flush_cb(
[... 166 lines cut ...]
                                hWnd,
                                uMsg,
                                wParam,
                                lParam,
                                plResult)) {
                        // Handled
                        return true;
                    }
                }

                // Not Handled
                return false;
            }
    }

    // Handled
    *plResult = 0;
    return true;
}

////////////////////////////////////////////////////////////////////////////////////////////

Let me know

nik

nikthefix, 2025-01-18:
@ou2356 

Hi there.

I deleted my previous message as it turns out my suggestion only worked on my win11 machine and failed on win10.

Maybe try this (which seems to work on both and is simpler):

In lvgl/src/drivers/windows/lv_windows_context.c

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

  case WM_SIZE: {
         // if(wParam != SIZE_MINIMIZED) {
              lv_windows_window_context_t * context = (lv_windows_window_context_t *)(
                                                          lv_windows_get_window_context(hWnd));
              if(context) {
                  if(!context->simulator_mode) {
                      context->display_resolution_changed = true;
                      context->requested_display_resolution.x = LOWORD(lParam);
                      context->requested_display_resolution.y = HIWORD(lParam);
                  }
                  else {
                      int32_t window_width = lv_windows_dpi_to_physical(
                                                 lv_windows_zoom_to_physical(
                                                     lv_display_get_horizontal_resolution(
                                                         context->display_device_object),
                                                     context->zoom_level),
                                                 context->window_dpi);
                      int32_t window_height = lv_windows_dpi_to_physical(
                                                  lv_windows_zoom_to_physical(
                                                      lv_display_get_vertical_resolution(
                                                          context->display_device_object),
                                                      context->zoom_level),
                                                  context->window_dpi);

                      RECT window_rect;
                      GetWindowRect(hWnd, &window_rect);

                      RECT client_rect;
                      GetClientRect(hWnd, &client_rect);

                      int32_t original_window_width =
                          window_rect.right - window_rect.left;
                      int32_t original_window_height =
                          window_rect.bottom - window_rect.top;

                      int32_t original_client_width =
                          client_rect.right - client_rect.left;
                      int32_t original_client_height =
                          client_rect.bottom - client_rect.top;

                      int32_t reserved_width =
                          original_window_width - original_client_width;
                      int32_t reserved_height =
                          original_window_height - original_client_height;

                      SetWindowPos(
                          hWnd,
                          NULL,
                          0,
                          0,
                          reserved_width + window_width,
                          reserved_height + window_height,
                          SWP_NOZORDER | SWP_NOACTIVATE | SWP_NOMOVE);               
                  }
              }

         // }
              
          break;
      
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

nik

ou2356, 2025-01-19:
@nikthefix 
Hi !
I commented out the condition 'if (wParam!=SIZE-MIIMIZED) {' based on your suggestion and found that it worked. 
```
        case WM_SIZE: {
                //if(wParam != SIZE_MINIMIZED) {
                    lv_windows_window_context_t * context = (lv_windows_window_context_t *)(
                                                                lv_windows_get_window_context(hWnd));
                    if(context) {
                        if(!context->simulator_mode) {
                            context->display_resolution_changed = true;
                            context->requested_display_resolution.x = LOWORD(lParam);
                            context->requested_display_resolution.y = HIWORD(lParam);
                        }
                        else {
                            int32_t window_width = lv_windows_dpi_to_physical(
                                                       lv_windows_zoom_to_physical(
                                                           lv_display_get_horizontal_resolution(
                                                               context->display_device_object),
                                                           context->zoom_level),
                                                       context->window_dpi);
                            int32_t window_height = lv_windows_dpi_to_physical(
                                                        lv_windows_zoom_to_physical(
                                                            lv_display_get_vertical_resolution(
                                                                context->display_device_object),
                                                            context->zoom_level),
                                                        context->window_dpi);

                            RECT window_rect;
                            GetWindowRect(hWnd, &window_rect);

                            RECT client_rect;
                            GetClientRect(hWnd, &client_rect);

                            int32_t original_window_width =
                                window_rect.right - window_rect.left;
                            int32_t original_window_height =
                                window_rect.bottom - window_rect.top;

                            int32_t original_client_width =
                                client_rect.right - client_rect.left;
                            int32_t original_client_height =
                                client_rect.bottom - client_rect.top;

                            int32_t reserved_width =
                                original_window_width - original_client_width;
                            int32_t reserved_height =
                                original_window_height - original_client_height;

                            SetWindowPos(
                                hWnd,
                                NULL,
                                0,
                                0,
                                reserved_width + window_width,
                                reserved_height + window_height,
                                SWP_NOZORDER | SWP_NOACTIVATE | SWP_NOMOVE);
                        }
                    }
                //}
                break;
            }
```
My testing environment is as follows:
Visual Studio 2022
win 11 x64
v9.0.0 release
LVGL v9.3.0

Thank you very much for your hard work! I am extremely grateful.

nikthefix, 2025-01-19:
@ou2356 
You're welcome and thanks for feeding back.