lvgl/lv_port_pc_visual_studio #73: Getting RLottie to work

tank104, 2024-05-28:
Has anyone been able to get rlottie to work?

I enabled LV_USE_RLOTTIE = 1

But get the error below.

That file is in https://github.com/Samsung/rlottie/tree/master/inc but not included in this, and I am unsure how I should best include it?

Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	C1083	Cannot open include file: 'rlottie_capi.h': No such file or directory	LvglWindowsStatic	C:\github\lv_port_pc_visual_studio\LvglPlatform\lvgl\src\libs\rlottie\lv_rlottie.c	12

kisvegabor, 2024-06-04:
Hi,

If you are using LVGL v9 I suggest using the new [Lottie library](https://docs.lvgl.io/master/widgets/lottie.html). It works with ThorVG which is integrated into LVGL.

tank104, 2024-06-07:
@kisvegabor thanks for this info - probably raises more questions than answers ;-)

Is there an example of this for the LV Port PC - I had a quick look and couldn't find anything.

tank104, 2024-06-11:
Ignore that - have found the examples - I had an older version of lvgl which is why I couldn't see them.

Got it all working now thanks!

closed 2024-06-11 by tank104

kisvegabor, 2024-06-13:
Fantastic! :slightly_smiling_face: