lvgl/lv_port_pc_visual_studio #74: Failed to locate "CL.exe"

focussing, 2024-06-07:
Hi!
I did exactly what is described in the readme, when I press the Local Windows debugger I get this message `Failed to locate "CL.exe"`
This is my Visual Studio info:

```
Microsoft Visual Studio Community 2022
Version 17.10.1
VisualStudio.17.Release/17.10.1+34928.147
Microsoft .NET Framework
Version 4.8.09037

Installed Version: Community

Visual C++ 2022   00482-90000-00000-AA286
Microsoft Visual C++ 2022

ASP.NET and Web Tools   17.10.338.1105
ASP.NET and Web Tools

Azure App Service Tools v3.0.0   17.10.338.1105
Azure App Service Tools v3.0.0

C# Tools   4.10.0-3.24270.2+e8f775c1d8a73dee7ad02408712d714251e708ea
C# components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

GitHub Copilot   0.2.889.30432
GitHub Copilot is an AI pair programmer that helps you write code faster and with less work.

Linux Core Dump Debugging   1.0.9.34902
Enables debugging of Linux core dumps.

Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines

NuGet Package Manager   6.10.0
NuGet Package Manager in Visual Studio. For more information about NuGet, visit https://docs.nuget.org/

TypeScript Tools   17.0.30327.2001
TypeScript Tools for Microsoft Visual Studio

Visual Basic Tools   4.10.0-3.24270.2+e8f775c1d8a73dee7ad02408712d714251e708ea
Visual Basic components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Visual C++ for Linux Development   1.0.9.34902
Visual C++ for Linux Development

Visual F# Tools   17.10.0-beta.24228.1+dd749058c91585e9b5dae62b0f8df892429ee28f
Microsoft Visual F# Tools

Visual Studio IntelliCode   2.2
AI-assisted development for Visual Studio.
```

MouriNaruto, 2024-10-09:
Close this issue due to no further replies.

Kenji Mouri

closed 2024-10-09 by MouriNaruto

focussing, 2024-10-14:
The proposed solution above using a CLI command is not working, because there is no `bin `directory in my `14.40.nnnnn` directory.
Also when I do a rebuild, the following additional message appears:

`Warning	MSB8003	The VCToolsInstallDir property is not defined. Some build tools may not be found.	LvglWindowsSimulator	C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets	512	`	

The issue is still present: I cannot get the source code via git, and rebuild as described in the readme.

Could you please provide a working code base?

MouriNaruto, 2024-10-14:
It will help to reinstall your Visual Studio toolchain if there is no bin folder in 14.40.xxxxx.

Kenji Mouri