lvgl/lv_port_pc_visual_studio #77: VS2022 v9.0.0 can't display img (lv_img_set_src(img, &picture); )

19e2k67, 2024-07-15:
[image]

But CodeBlocks lvgl9.1 can display images normally！

[image]

laishanhai1040, 2024-09-23:
I also encountered the same problem. The examples and demos were normal, but the code I wrote myself did not work.

laishanhai1040, 2024-09-24:
我发现这是 C 和 C++ 语法不同的原因，经过测试，可以为 lvgl 生成的 img_test.c 添加一个 img_test.h 文件，并且添加上
`#ifdef __cplusplus
extern "C" {
#endif
extern const lv_image_dsc_t img_test;
#ifdef __cplusplus
} /*extern "C"*/
#endif`

在我的电脑上测试是可以正常显示的。

laishanhai1040, 2024-09-24:
我发现这是 C 和 C++ 语法不同的原因，经过测试，可以为 lvgl 生成的 img_test.c 添加一个 img_test.h 文件，并且添加上
'''#ifdef __cplusplus
extern "C" { 
#endif

extern const lv_image_dsc_t img_test; 

#ifdef __cplusplus 
} /*extern "C"*/ 
#endif '''

在我的电脑上测试是可以正常显示的。