lvgl/lv_port_pc_visual_studio #80: Not Compiling With VS 2019 or 2022

vwheeler63, 2024-09-11:
I did a fresh clone of `lv_port_pc_visual_studio` and tried to open it in Visual Studio 2022, and I get this message in the middle of the display area of the file LvglWindowsSimulator.vcxproj:

```
An error occurred in 'LvglWindowsSimulator' while attempting to open 'LvglWindowsSimulator.vcxproj'
Value does not fall within the expected range.
```
and this which it would not let me copy/paste so I got a screen shot:

[image]

Trying to build the project results in errors from missing files:
- ./src/drivers/display/renesas_glcdc/lv_renesas_glcdc.c
- ./src/drivers/glfw/lv_glfw_mouse.c
- ./src/drivers/glfw/lv_glfw_mouse_private.c
- ./src/drivers/glfw/lv_glfw_window.c
- ./src/drivers/glfw/lv_opengles_debug.c
- ./src/drivers/glfw/lv_opengles_driver.c
- ./src/drivers/libinput/lv_libinput.c
- ./src/drivers/libinput/lv_xkb.c

...and several others -- too many to try to list (estimate based on scanning the error list with my eyes, 1306 errors and warnings).

I tried changing the `lvgl` submodule's "current state" to:
- release/v8.4 branch
- v9.0.0 tag
- v9.1.0 tag
- v9.2.0 tag, and
- master branch

but none of them handled this problem.  In fact, I cannot find any record of the `lv_glfw_mouse*.c` files existing in the recent `lvgl/lvgl` repository.  Are they pending in a PR that has not been merged yet?

=========

Next, leaving  the `lvgl` submodule on the `v9.2.0` tag, I tried it again with Visual Studio 2019.  I don't get the error opening the project as I do with VS 2022.  But attempting to do a build (from scratch after a `clean`) ends very quickly with the following errors:

```
Severity	Code	Description	Project	File	Line	Suppression State
Warning	LTL2003	VC-LTL is disabled because VC-LTL isn't compatible with the  toolset. Please switch to Visual Studio 2015/2017/2019/2022 and continue.	E:\Dev\Clients\WGA\lv_port_pc_visual_studio\LvglWindows\LvglWindowsStatic.vcxproj	C:\Users\Vic\.nuget\packages\vc-ltl\5.1.1\build\native\config\config.props	155	

    <Warning Code="LTL2003" Text="$(ERROR_VC_LTL_NOT_SUPPORT_PLATFORM_TOOLSET)" Condition="'$(VCLTLSupportPlatformToolset)' !='true'"/>

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB4018	The "VCMessage" task failed unexpectedly.
System.FormatException: Index (zero based) must be greater than or equal to zero and less than the size of the argument list.
   at System.Text.StringBuilder.AppendFormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.FormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.Format(IFormatProvider provider, String format, Object[] args)
   at Microsoft.Build.Shared.ResourceUtilities.FormatString(String unformatted, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.FormatString(String unformatted, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.FormatResourceString(String resourceName, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.LogErrorWithCodeFromResources(String messageResourceName, Object[] messageArgs)
   at Microsoft.Build.CPPTasks.VCMessage.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	LvglWindows	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets	439	

Severity	Code	Description	Project	File	Line	Suppression State
Warning	LTL2003	VC-LTL is disabled because VC-LTL isn't compatible with the  toolset. Please switch to Visual Studio 2015/2017/2019/2022 and continue.	LvglWindowsDesktopApplication	C:\Users\Vic\.nuget\packages\vc-ltl\5.1.1\build\native\config\config.props	155	

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB4018	The "VCMessage" task failed unexpectedly.
System.FormatException: Index (zero based) must be greater than or equal to zero and less than the size of the argument list.
   at System.Text.StringBuilder.AppendFormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.FormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.Format(IFormatProvider provider, String format, Object[] args)
   at Microsoft.Build.Shared.ResourceUtilities.FormatString(String unformatted, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.FormatString(String unformatted, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.FormatResourceString(String resourceName, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.LogErrorWithCodeFromResources(String messageResourceName, Object[] messageArgs)
   at Microsoft.Build.CPPTasks.VCMessage.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	LvglWindowsDesktopApplication	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets	439	

Severity	Code	Description	Project	File	Line	Suppression State
Warning	LTL2003	VC-LTL is disabled because VC-LTL isn't compatible with the  toolset. Please switch to Visual Studio 2015/2017/2019/2022 and continue.	LvglWindowsSimulator	C:\Users\Vic\.nuget\packages\vc-ltl\5.1.1\build\native\config\config.props	155	

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB4018	The "VCMessage" task failed unexpectedly.
System.FormatException: Index (zero based) must be greater than or equal to zero and less than the size of the argument list.
   at System.Text.StringBuilder.AppendFormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.FormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.Format(IFormatProvider provider, String format, Object[] args)
   at Microsoft.Build.Shared.ResourceUtilities.FormatString(String unformatted, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.FormatString(String unformatted, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.FormatResourceString(String resourceName, Object[] args)
   at Microsoft.Build.Utilities.TaskLoggingHelper.LogErrorWithCodeFromResources(String messageResourceName, Object[] messageArgs)
   at Microsoft.Build.CPPTasks.VCMessage.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	LvglWindowsSimulator	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets	439	

```

This project has been really important for my last project, and I would hate to think that others are missing out.

I kept a copy of the version of this project I was working with, so it is not currently broken, but I thought  you should be aware of this if you were not already.

Kind regards,
Vic

kisvegabor, 2024-09-12:
This one might be required: https://github.com/lvgl/lv_port_pc_visual_studio/blob/master/Documents/HowToSynchronizeLvglRelatedSubmodules.md

In Eclipse I can just right click on the project and click "refresh" to get the new files. Isn't there a similarly simple way VS too?

vwheeler63, 2024-09-12:
> This one might be required: https://github.com/lvgl/lv_port_pc_visual_studio/blob/master/Documents/HowToSynchronizeLvglRelatedSubmodules.md
> 
> In Eclipse I can just right click on the project and click "refresh" to get the new files. Isn't there a similarly simple way VS too?

In VS 2022, adding and removing files is is automatic by adding or removing them from the project directory (similar to PyCharm).  In VS 2019, there is a fast way (hidden) way to add existing files, explained in this [stackoverflow article](https://stackoverflow.com/questions/20033220/visual-studio-add-existing-folders-to-project), and removing them is still manual, though easy (by right click).

Thinking about it, if I could get VS 2022 past that error message (early in the description), then it would probably handle the missing/moved files automatically.  If one approaches it with VS 2019, it can be faster to edit the `.vcxproj` file and change the paths.  Especially if there are several of them.

MouriNaruto, 2024-09-12:
Visual Studio C++ projects will note every file in the project configuration file. So, it need to use a tool to update.

Kenji Mouri

vwheeler63, 2024-09-12:
> Visual Studio C++ projects will note every file in the project configuration file. So, it need to use a tool to update.

Ah....  Indeed my test with VS 2022 was with a VB.NET project.

vwheeler63, 2024-10-01:
I see that it compiles now as of the commits on 16-Sep-2024.  Thank you, @MouriNaruto !

closed 2024-10-01 by vwheeler63

vwheeler63, 2024-11-20:
Hi, @MouriNaruto !

I notice that this project has the directory structure preserved in the project, as opposed to Visual Studio's default of having all `.c` files in one "folder" and all `.h` files in another "folder" (which I really don't like for large projects).  Do you have a utility program of some sort that modifies the `.cproj` file to do that?  If so, I would very much like to get it, since I am trying to maintain a (private) Visual Studio project that ONLY compiles the LVGL library as a compilation test before I submit code changes.

Would you mind sharing how you make `lvgl/lv_port_pc_visual_studio` project so useful by preserving the directory structure?

MouriNaruto, 2024-11-20:
> Hi, @MouriNaruto !
> 
> I notice that this project has the directory structure is (very usefully) preserved in the project, as opposed to Visual Studio's default of having all `.c` files in one "folder" and all `.h` files in another "folder" (which I really don't like for large projects). Do you have a utility program of some sort that modifies the `.cproj` file to do that? If so, I would very much like to get it, since I am trying to maintain a (private) Visual Studio project that ONLY compiles the LVGL library as a compilation test before I submit code changes.
> 
> Would you mind sharing how you make `lvgl/lv_port_pc_visual_studio` project so useful by preserving the directory structure?

The source code of the project definition generator is in this repository: https://github.com/lvgl/lv_port_pc_visual_studio/tree/master/LvglProjectFileUpdater

Kenji Mouri

vwheeler63, 2024-11-20:
> The source code of the project definition generator is in this repository: https://github.com/lvgl/lv_port_pc_visual_studio/tree/master/LvglProjectFileUpdater

Excellent!!  Thank you very much!  😄