lvgl/lv_port_pc_visual_studio #83: VS 2022 V9.3.0 The widget I wrote with reference to lv_objx_template cannot be displayed

Nickyrat, 2024-10-17:
/**
 * @file lv_mybtn.h
 *
 */

 /**
  * TODO Remove these instructions
  * Search and replace: mybtn -> object short name with lower case(e.g. btn, label etc)
  *                    MYBTN -> object short name with upper case (e.g. BTN, LABEL etc.)
  *
  */

#ifndef LV_MYBTN_H
#define LV_MYBTN_H

#ifdef __cplusplus
extern "C" {
#endif

    /*********************
     *      INCLUDES
     *********************/
#include "lvgl/src/lv_conf_internal.h"

#if LV_USE_MYBTN != 0

#include "lvgl/src/core/lv_obj.h"

     /*********************
      *      DEFINES
      *********************/

      /**********************
       *      TYPEDEFS
       **********************/
       /*Data of mybtnate*/
    typedef struct {
        lv_obj_t obj;
        lv_style_t style;
        bool pressed;
        lv_color_t backgroundColor;
        char* text;
    } lv_mybtn_t;

    extern const lv_obj_class_t lv_mybtn_class;

    /**********************
     * GLOBAL PROTOTYPES
     **********************/

     /**
      * Create a mybtn object
      * @param parent    pointer to an object, it will be the parent of the new mybtn
      * @return          pointer to the created bar
      */
    lv_obj_t* lv_mybtn_create(lv_obj_t* parent);

    /*======================
     * Add/remove functions
     *=====================*/

     /*=====================
      * Setter functions
      *====================*/

      /*=====================
       * Getter functions
       *====================*/

       /*=====================
        * Other functions
        *====================*/

    void lv_pushbutton_set_text(lv_obj_t* obj, const char* text);

    /**********************
     *      MACROS
     **********************/

#endif /*LV_USE_MYBTN*/

#ifdef __cplusplus
} /*extern "C"*/
#endif

#endif /*LV_MYBTN_H*/

/**
 * @file lv_mybtn.c
 *
 */

 /**
  * TODO Remove these instructions
  * Search and replace: mybtn -> object short name with lower case(e.g. btn, label etc)
  *                    MYBTN -> object short name with upper case (e.g. BTN, LABEL etc.)
  *
  * You can remove the defined() clause from the #if statement below. This exists because
  * LV_USE_MYBTN is not in lv_conf.h or lv_conf_mybtnate.h by default.
  */

  /*********************
   *      INCLUDES
   *********************/
#include "lv_mybtn.h" /*TODO uncomment this*/

#if LV_USE_MYBTN != 0

   /*********************
    *      DEFINES
    *********************/
#define MY_CLASS &lv_mybtn_class

    /**********************
     *      TYPEDEFS
     **********************/

     /**********************
      *  STATIC PROTOTYPES
      **********************/
static void lv_mybtn_constructor(const lv_obj_class_t* class_p, lv_obj_t* obj);
static void lv_mybtn_destructor(const lv_obj_class_t* class_p, lv_obj_t* obj);
static void lv_mybtn_event(const lv_obj_class_t* class_p, lv_event_t* e);

/**********************
 *  STATIC VARIABLES
 **********************/
const lv_obj_class_t lv_mybtn_class = {
    .constructor_cb = lv_mybtn_constructor,
    .destructor_cb = lv_mybtn_destructor,
    .event_cb = lv_mybtn_event,
    .width_def = LV_DPI_DEF,
    .height_def = LV_DPI_DEF,
    .instance_size = sizeof(lv_mybtn_t),
    .group_def = LV_OBJ_CLASS_GROUP_DEF_INHERIT,
    .editable = LV_OBJ_CLASS_EDITABLE_INHERIT,
    .base_class = &lv_mybtn_class
};

/**********************
 *      MACROS
 **********************/

 /**********************
  *   GLOBAL FUNCTIONS
  **********************/

lv_obj_t* lv_mybtn_create(lv_obj_t* parent)
{
    LV_LOG_INFO("begin");
    lv_obj_t* obj = lv_obj_class_create_obj(MY_CLASS, parent);
    lv_obj_class_init_obj(obj);
    return obj;
}

/*======================
 * Add/remove functions
 *=====================*/

 /*
  * New object specific "add" or "remove" functions come here
  */

  /*=====================
   * Setter functions
   *====================*/

   /*
    * New object specific "set" functions come here
    */

    /*=====================
     * Getter functions
     *====================*/

     /*
      * New object specific "get" functions come here
      */

      /*=====================
       * Other functions
       *====================*/

       /*
        * New object specific "other" functions come here
        */

        /**********************
         *   STATIC FUNCTIONS
         **********************/

static void lv_mybtn_constructor(const lv_obj_class_t* class_p, lv_obj_t* obj)
{
    LV_UNUSED(class_p);
    LV_TRACE_OBJ_CREATE("begin");

    //lv_mybtn_t* mybtn = (lv_mybtn_t*)obj;
    /*Initialize the widget's data*/
    lv_mybtn_t* btn = (lv_mybtn_t*)obj;
    btn->pressed = false;
    btn->text = "OK";

    lv_style_t style = btn->style;
    lv_style_init(&style);
    lv_style_set_bg_opa(&style, LV_OPA_COVER);//必不可少，否则默认控件背景不可见

    lv_style_set_bg_color(&style, lv_color_hex(0xe8e8e8));
    lv_style_set_text_font(&style, &lv_font_montserrat_8);
    lv_style_set_radius(&style, 12);
    lv_style_set_border_width(&style, 10);
    lv_style_set_border_color(&style, lv_color_hex(0x33dd33));
    lv_style_set_border_opa(&style, LV_OPA_50);

    lv_obj_add_style(obj, &style, LV_PART_MAIN);

    LV_TRACE_OBJ_CREATE("finished");
}

static void lv_mybtn_destructor(const lv_obj_class_t* class_p, lv_obj_t* obj)
{
    LV_UNUSED(class_p);
    lv_mybtn_t* mybtn = (lv_mybtn_t*)obj;
    /*Free the widget specific data*/
}

static void lv_mybtn_event(const lv_obj_class_t* class_p, lv_event_t* e)
{
    LV_UNUSED(class_p);

    //lv_res_t res;

    ///*Call the ancestor's event handler*/
    //res = LV_EVENT_base(MY_CLASS, e);
    //if (res != LV_RES_OK) return;

    /*Add the widget specific event handling here*/

    lv_res_t res = lv_obj_event_base(MY_CLASS, e);
    if (res != LV_RES_OK)
        return;

    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t* obj = lv_event_get_target(e);

    if (code == LV_EVENT_REFR_EXT_DRAW_SIZE) {
        lv_coord_t* s = (lv_coord_t*)lv_event_get_param(e);
        *s = LV_MAX(*s, 20);
    }
    else if (code == LV_EVENT_DRAW_MAIN)
    {
        lv_mybtn_t* btn = (lv_mybtn_t*)obj;

        lv_draw_rect_dsc_t rect_dsc;
        lv_draw_rect_dsc_init(&rect_dsc);
        lv_obj_init_draw_rect_dsc(obj, LV_PART_MAIN, &rect_dsc);//用obj的 LV_PART_MAIN 样式属性初始化 rect_dsc

        if (btn->pressed)
        {
            rect_dsc.bg_color = lv_color_mix(rect_dsc.bg_color, lv_color_black(), 200);
        }

        const lv_area_t* clip_area = (const lv_area_t*)lv_event_get_param(e);

        const lv_area_t area = obj->coords;

        lv_obj_draw_part_dsc_t  part_draw_dsc;
        lv_obj_draw_dsc_init(&part_draw_dsc, clip_area);
        part_draw_dsc.class_p = &lv_obj_class;
        part_draw_dsc.rect_dsc = &rect_dsc;
        part_draw_dsc.part = LV_PART_MAIN;

        lv_draw_rect(&area, clip_area, &rect_dsc);

        const char* text = btn->text;

        lv_point_t txt_size;
        lv_txt_get_size(&txt_size, text,
            lv_obj_get_style_text_font(obj, LV_PART_MAIN),
            lv_obj_get_style_text_letter_space(obj, LV_PART_MAIN),
            0,
            LV_COORD_MAX, LV_TEXT_FLAG_NONE);

        lv_area_t txt_coords;
        txt_coords.x1 = area.x1 + (area.x2 - area.x1 - txt_size.x) / 2;
        txt_coords.y1 = area.y1 + (area.y2 - area.y1 - txt_size.y) / 2;
        txt_coords.x2 = txt_coords.x1 + txt_size.x;
        txt_coords.y2 = txt_coords.y1 + txt_size.y;

        lv_draw_label_dsc_t label_dsc;
        lv_draw_label_dsc_init(&label_dsc);
        lv_obj_init_draw_label_dsc(obj, LV_PART_MAIN, &label_dsc);

        lv_draw_label(&txt_coords, clip_area, &label_dsc, text, NULL);
    }
    else if (code == LV_EVENT_PRESSED)
    {
        lv_mybtn_t* btn = (lv_mybtn_t*)obj;
        btn->pressed = true;
        lv_obj_invalidate(obj);
    }
    else if (code == LV_EVENT_RELEASED)
    {
        lv_mybtn_t* btn = (lv_mybtn_t*)obj;
        btn->pressed = false;
        lv_obj_invalidate(obj);
    }
}

void lv_pushbutton_set_text(lv_obj_t* obj, const char* text)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);
    lv_mybtn_t* btn = (lv_mybtn_t*)obj;
    btn->text = text;
}

#else /*Enable this file at the top*/

   /*This dummy typedef exists purely to silence -Wpedantic.*/
typedef int keep_pedantic_happy;
#endif

I added these two files in LvglWindowsSimulator and added the macro LV_USE_MYBTN in lv_conf. h, but it doesn't seem to be very useful? What is the correct approach?

kisvegabor, 2024-10-17:
Please format the code like
````
```c
my
code
here
```
````

Nickyrat, 2024-10-30:
resolved

closed 2024-10-30 by Nickyrat