lvgl/lv_port_pc_visual_studio #84: Rendering issues

kisvegabor, 2024-10-21:
Hi,

It was reported by multiple people that there are some rendering issues. Especially with the line and chart example. All these were related to the Visual Studio project so I believe it's not generic LVGL issue but rather a Windows driver issue of some kind. 

The last reported issue is here: https://forum.lvgl.io/t/two-questions-about-the-chart-control/18590/2

MouriNaruto, 2024-10-21:
OK. I will try to reproduce for that.

Kenji Mouri

MouriNaruto, 2024-10-21:
@kisvegabor 

[image]

I solve the issue when I comment out all multi-thread rendering logic for LVGL software renderer. It looks the the string buffer from event_cb in lv_example_chart_3 will not transfer to the render thread properly.

Kenji Mouri

kisvegabor, 2024-10-24:
Thank you for looking into it. As temporal solution let's disable multi threading.

MouriNaruto, 2024-10-24:
@kisvegabor But it seems no options for disabling multi-threading rendering support without setting LV_USE_OS to none. (Windows backend needs to set LV_USE_OS to Windows.)

Kenji Mouri

kisvegabor, 2024-10-24:
> Windows backend needs to set LV_USE_OS to Windows

Why is this?

MouriNaruto, 2024-10-24:
> > Windows backend needs to set LV_USE_OS to Windows
> 
> Why is this?

Because Windows backend need to use multi-thread support to split the windows thread and LVGL main thread. (For making LVGL main thread will not stuck when people moving the windows.)

Kenji Mouri

kisvegabor, 2024-10-25:
Can you use normal Windows mutex for it instead of `lv_mutex`? If so, LVGL's OS could be disabled.

MouriNaruto, 2024-10-27:
> Can you use normal Windows mutex for it instead of `lv_mutex`? If so, LVGL's OS could be disabled.

I can implement the Windows backend with the standalone mutex implementation. But in my opinion, it may not be a good idea. (some people may hope to use LVGL‘s OS Abstraction Layer without the multi-thread LVGL SW rendering)

Kenji Mouri

kisvegabor, 2024-10-27:
I agree, I just meant it as a hot fix. Do you have any idea what the problem could be? Can you run git bisect to find the offending commit?

MouriNaruto, 2024-10-28:
> I agree, I just meant it as a hot fix. Do you have any idea what the problem could be? Can you run git bisect to find the offending commit?

Because the buffer is invalid when LVGL's software render queue executing the task.

It will be fixed if I change the following line to `static char buf[16];`
https://github.com/lvgl/lvgl/blob/c27760fb103dbcee94bbc2751e741fb55ffa8f35/examples/widgets/chart/lv_example_chart_3.c#L30

But I don't think it's a good fix. May @kisvegabor should handle the buffer properly in lv_draw_rect or other functions.

Kenji Mouri

MouriNaruto, 2024-10-28:
> Ahh, it's a bug in the example. LVGL assumes the the image source is always static, but now it's a local variable and we save only its pointer. I can fix it in the example.
> 
> Could you take a look at the line related examples too? Some errors were reported for them too.

https://github.com/lvgl/lvgl/blob/c27760fb103dbcee94bbc2751e741fb55ffa8f35/src/draw/sw/lv_draw_sw_mask_private.h#L56
Change lv_draw_sw_mask_line_side_t to uint8_t because lv_draw_sw_mask_line_side_t will be seen as signed int. It actually causes LV_DRAW_SW_MASK_LINE_SIDE_TOP = -2 and LV_DRAW_SW_MASK_LINE_SIDE_BOTTOM = -3;

Kenji Mouri

kisvegabor, 2024-10-28:
Thank you, Kenji! 💯 
I've just open a PR to fix both: https://github.com/lvgl/lvgl/pull/7171

kisvegabor, 2024-10-29:
As the PR is merged I'm closing this issue. Thank you again!