lvgl/lv_port_pc_visual_studio #88: VS2022 : The project could not be loaded - The SDK 'Mile.Project.Configurations/1.0.1426' specified could not be found.

mallyhubz, 2024-11-11:
VS2022 : The project could not be loaded -

 The SDK 'Mile.Project.Configurations/1.0.1426' specified could not be found.

MouriNaruto, 2024-11-12:
Does your environment connect to NuGet properly?

Kenji Mouri

VicRoss, 2024-11-23:
I am getting the same error along with a few more, all related to a missing SDK.  I don't understand what NuGet is or how to connect to it or why I should have to.

I have cloned the repository several times using bot the All at Once and one at a time method to insure the sub repositories are installed.

I have reinstalled Visual Studio several times as well.  Nothing works to get rid of these errors.

Poing3000, 2025-02-26:
I had the same issue and solved it by manually adding _https://api.nuget.org/v3/index.json_ as a source to the Package Sources in Visual.

1. In Visual Studio, go to _Tools → Options → NuGet Package Manager → Package Sources_.
2. Click on the + at the top right and change the source at the bottom to: _https://api.nuget.org/v3/index.json_

Not sure if this is the best way; though it works.

benzhiqiang, 2025-03-11:
> I had the same issue and solved it by manually adding _https://api.nuget.org/v3/index.json_ as a source to the Package Sources in Visual.
>
>     1. In Visual Studio, go to _Tools → Options → NuGet [...]

thankyou 20250311，your method works