lvgl/lv_port_pc_visual_studio #101: 2025 Roadmap for LVGL Visual Studio repository

MouriNaruto, 2025-03-25:
I think I should back to the typical contributions for LVGL Visual Studio repository again after I proceed several personal projects. Also, LVGL mainline's new UEFI support makes me excited.

Here is the list I will try my best to finish in 2025:

- [ ] Add custom MSBuild task for synchronizing the Visual Studio project automatically after updating the LVGL submodule. (For resolving the issue in https://github.com/lvgl/lvgl/issues/7412)
  - [ ] Visual Studio project configuration synchronization support.
  - [ ] Windows dynamic library binary export definition file synchronization support.
  - [ ] lv_conf.h synchronization support. (I need to do some researches for that.)
- [ ] Visual Studio UEFI project example.
  - [ ] Prepare the lightweight UEFI specification headers which distributed via NuGet. (Use dependencies without submodule as possible will simplify the maintenance. Here is the work-in-progress project which I have used in my projects since 2021: https://github.com/ProjectMile/Mile.Uefi)
- [ ] Visual Studio Hyper-V Generation 2 Virtual Machines bare-metal project example. (I planned for that since 2023.)
  - [x] Prepare the interface headers which distributed via NuGet. (Ready, read https://github.com/ProjectMile/Mile.HyperV for the header definitions.)
  - [ ] Prepare the lightweight UEFI specification headers which distributed via NuGet. (Because Hyper-V Generation 2 Virtual Machines only provide the 64-Bit UEFI Class 3 firmware.)
  - [ ] Write the VMBus driver for LVGL and hope to make it merged in the mainline.
- [ ] Improve the documents for repository. (Current documents for this repository is not friendly.)
  - [ ] Add more documents to explain what and why of some issues.

Kenji Mouri

kisvegabor, 2025-03-25:
Hey, 

happy that you are back again!  

> Add custom MSBuild task for synchronizing the Visual Studio project automatically after updating the LVGL submodule. (For resolving the issue in https://github.com/lvgl/lvgl/issues/7412)

It's a very important one. If it'd work automatically (just like in Eclipse or VSCode) I could recommend this projects for more users. 

> lv_conf.h synchronization support. (I need to do some researches for that.)

I suggest looking into [this](https://github.com/lvgl/lvgl/blob/master/scripts/generate_lv_conf.py). It can generate `lv_conf.h` from `lv_conf.defaults` (like [this](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/lv_conf.defaults))

MouriNaruto, 2025-04-07:
@kisvegabor 

> I suggest looking into [this](https://github.com/lvgl/lvgl/blob/master/scripts/generate_lv_conf.py). It can generate `lv_conf.h` from `lv_conf.defaults` (like [this](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/lv_conf.defaults))

I tried to make a lv_conf.defaults compatible generator for workaround that.

Kenji Mouri

vwheeler63, 2025-05-19:
Not stale.

MouriNaruto, 2026-09-27:
Finally, I have done the "Add custom MSBuild task for synchronizing the Visual Studio project automatically after updating the LVGL submodule."

Parsing the user header file is too hard, so I follow the lv_conf.defaults.

Read https://github.com/lvgl/lv_port_pc_visual_studio/blob/master/Documents/HowToSynchronizeLvglRelatedSubmodules.md for more information.

Kenji Mouri